MarlinFirmware/Marlin #21846: [BUG] Marlin firmware.bin unable to load a second time in VScode for version 2.0.8

ebraiman, 2021-05-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

So when loading Marlin version 2.0.8 firmware.bin on SKR_v14 or MKS_Sgen_L_v2, firmware reload fails via USB.

### Bug Timeline

2.0.8

### Expected behavior

firmware.bin from VScode loads via serial(UBS) connection.

### Actual behavior

Firmware.bin fails to load from Marlin firmware upload in VScode for version 2.0.8 for both SKR_v14 (Turbo) and MKS sGen  L v2.0.

### Steps to Reproduce

1. download newest marlin firmware 2.0.8
2. open in VScode and extracting zip file for marlin firmware
3. change configuration.h for motherboard for either SKR_v14 or MKS_sgen_Lv2 or watch livestream video.
4. change platformio.ini for LPC1769
5.  Upload build to either motherboard for firmware.bin

### Version of Marlin Firmware

2.0.8

### Printer model

N/A 

### Electronics

SKR_14(turbo), MKS sGen L v2

### Add-ons

N/A

### Your Slicer

Other (explain below)

### Additional information & file uploads

[Uploading Marlin-2.0.8and2.0.7.2.zip…]()

ebraiman, 2021-05-08:
Here is what occurs at end of compile on VScode...

Upload disk:  C:\ 

Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed
CURRENT: upload_protocol = mbed
Looking for upload disk...
*** [upload] C:\firmware.bin: Permission denied
Use manually specified: C:\
Uploading .pio\build\LPC1769\firmware.bin

thisiskeithb, 2021-05-08:
This isn't a Marlin bug. Copy `firmware.bin` to your microSD card manually and place in your board in power cycle it.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-05-08 by thisiskeithb

ebraiman, 2021-05-08:
I understand what you're saying, but you've lost functionality between builds.  One would think this a bug?

ellensp, 2021-05-09:
The only way I can see this could happen is if your have a FIRMWARE.CUR in your C:\ This would really confused the uploader as one of the ways it locates the sd card drive is to look for FIRMWARE.CUR in the root of the drive

thisiskeithb, 2021-05-09:
> I understand what you're saying, but you've lost functionality between builds. One would think this a bug?

The "Upload" function is typically meant for a programmer like an ST-Link, J-Link, etc. PlatformIO will look for a `firmware.cur` file if your board exposes the SD card to the OS, but is not a reliable way to copy firmware to your board.

This is why I explained that you need to copy the file manually.

This is not a Marlin bug.

Please use the links in my last reply for further discussion since this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here.

ebraiman, 2021-05-09:
Watch the end of the tutorial here and see the functionality.

https://youtu.be/MbCe-dfTNAY  (This is in Marlin 2.0.7.2, try the same with 2.0.8)

ebraiman, 2021-05-09:
remember to try and load firmware a second time from VScode after first power-cycle in 2.0.8

ebraiman, 2021-05-09:
BTW, it's FIRMWARE.CUR and binary file is called firmware.bin.

ebraiman, 2021-05-09:
FYI, which link in Discord Marlin?  I posted a comment 2 hours ago?

ellensp, 2021-05-09:
"this functionality has existed since SKR 1.3 " This is not accurate. This functionally was added for REARM, which predates skr1.3 by quiet some time. This is why the uploader also looks for the volume label "REARM" (on Linux system)

I just tried this, multiple time on windows. I cannot get it to fail.

As I said before I suspect you have a FIRMWARE.CUR file in the root of your C drive. Unless you have a very odd system C drive is NOT your sd card. Delete FIRMWARE.CUR from your C drive root.

ebraiman, 2021-05-09:
You are correct the was a FIRMWARE.CUR on root of c: drive.  So here is the question, if it works with 2.0.7.2, why does it not with 2.0.8?