MarlinFirmware/Marlin #21852: [BUG] M33 ("Get Long Path") crashes firmware if no parameter provided

theomega, 2021-05-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The M33 command, according to the [documentation](https://marlinfw.org/docs/gcode/M033.html) requires a parameter. If this parameter is provided, the command works fine.

BUT: If you forget the provide the parameter, the firmware crashes.

This seems to be a null pointer the [source code](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/gcode/sd/M33.cpp) assumes that the parameter is there and then most likely crashes.

### Bug Timeline

Should have been the case since the M33 Command was introduced.

### Expected behavior

Error message is printed.

### Actual behavior

Firmware crashes

### Steps to Reproduce

1. Have your firmware compiled with `LONG_FILENAME_HOST_SUPPORT` (otherwise, there is no M33 command)
2. Execute `M33`

### Version of Marlin Firmware

2.0.8

### Printer model

Home Built

### Electronics

FYSETC E4 board, based on ESP32

XDA-Bam, 2021-05-12:
Is something like

```c++
void GcodeSuite::M33() {

  if(parser.string_arg)
    card.printLongPath(parser.string_arg);
  else
    SERIAL_ECHOLNPGM(STR_SD_CANT_OPEN_SUBDIR);

}
```

enough? Or do we need to define a new warning message?

charlespick, 2021-05-19:
> Is something like
> 
> ```c++
> void GcodeSuite::M33() {
> 
>   if(parser.string_arg)
>     card.printLongPath(parser.string_arg);
>   else
>     SERIAL_ECHOLNPGM(STR_SD_CANT_OPEN_SUBDIR);
> 
> }
> ```
> 
> enough? Or do we need to define a new warning message?

Looks like it's worth testing!

XDA-Bam, 2021-07-20:
Bad bot. I know how to fix it, but didn't get to it yet. Please keep this open.

thisiskeithb, 2021-07-20:
I cannot replicate the crashing using today's `bugfix-2.0.x` (3fdf40f), so please download a fresh copy of [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

sjasonsmith, 2021-11-27:
This crash is now eliminated, by eliminating a `strlen` call with a NULL pointer. It does not print any error message, just a blank line of output for the "long" filename. There could be other ways to resolve this (as discussed above), but since this is an incorrect use-case anyway, I think that just preventing crashing should be good enough.

closed 2021-11-27 by sjasonsmith