MarlinFirmware/Marlin #21853: #define NO_MOTION_BEFORE_HOMING is not working as expected and was working in 2.0.7.x builds

CBDesignS, 2021-05-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

#define NO_MOTION_BEFORE_HOMING in both bugfix and stable branches of Marlin 2.0.8.x lowers the build plate when enabled or dsabled. this crunches the stepper as the build plate lowers away from nozel at the end of the build. when the next print run is started and a home all axis is initiated the bug kicks in trying to move the Z axis seeming to ignore the NO_MOTION_BEFORE_HOMING. this was working great on 2.0.7 builds 

### Bug Timeline

that start of 2.0.8.x

### Expected behavior

build platform remains stationary

### Actual behavior

build platform lowers and crunches stepper because it is at max travel on the axis already.

### Steps to Reproduce

G28 to initiate home all axis. any thing that causes a home on the Z axis.

### Version of Marlin Firmware

2.0.8.x bugfix and stable

### Printer model

ender 5 with btt skr 1.4t & ctc dual with mks sgen-l v1

### Electronics

ender 5 with btt skr 1.4t & ctc dual with mks sgen-l v1

### Add-ons

ender 5 with inductive probe & ctc dual with 3d touch probe

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6447522/configs.zip)

CBDesignS, 2021-07-09:
As of build 2.0.9.1 bugfix this bug is still active.

CBDesignS, 2021-09-08:
> 
> 
> As of build 2.0.9.1 bugfix this bug is still active.

and another 60 days has gone by without anything done with this bug that will effect quite a lot of diving/spring board style printers that lower the bed instead of raising the print head, but I suppose adding new machine boards and fixing chinese mfr board problems is better for marlin than fixing old bugs. I know the devs are over stressed and over worked but at some point someone has to stand up and say enough is enough with all these new boards and start fixing all these outstanding bugs..

ellensp, 2021-09-08:
Take the  high road and ignoring all the negative commentary above. 

In Marlin/src/gcode/calibrate/G28.cpp  is   (line 371 in my version)
```CPP
    if (z_homing_height && (LINEAR_AXIS_GANG(doX, || doY, || TERN0(Z_SAFE_HOMING, doZ), || doI, || doJ, || doK))) {
      // Raise Z before homing any other axes and z is not already high enough (never lower z)
      if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPAIR("Raise Z (before homing) by ", z_homing_height);
      do_z_clearance(z_homing_height);
      TERN_(BLTOUCH, bltouch.init());
    }
```
Change this to

```CPP
    if (z_homing_height && (LINEAR_AXIS_GANG(doX, || doY, || TERN0(Z_SAFE_HOMING, doZ), || doI, || doJ, || doK))) {
      #if DISABLED(NO_MOTION_BEFORE_HOMING)
        // Raise Z before homing any other axes and z is not already high enough (never lower z)
        if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPAIR("Raise Z (before homing) by ", z_homing_height);
        do_z_clearance(z_homing_height);
      #endif
      TERN_(BLTOUCH, bltouch.init());
    }
```

And its sorted.

CBDesignS, 2021-09-08:
I am sorry If I come over a tad pushy but if you managed to get the fix to this problem within a couple of mins of me having a rant could this not have been fixed a long time ago @ellensp ? or even a note saying that a fix may have been found but needs testing ? NO it is just left to expire and be auto closed by the bot. I worked arround the bug by making my build area 30mm smaller to allow this bug to happen.. Alas I am that stupid I do not know what the hell to do with what you have posted above but thankyou for your reply. I now know my place in the pecking list of thick idiots and I will never darken the pages of this git again. Thankyou for all you hard work

closed 2021-09-08 by CBDesignS

CBDesignS, 2021-09-08:
this fix posted by @ellensp does as it says, it stops the bed from moving while X & Y axis home. I know the steppers have no idea where they are at machine power up and marlin will poss do some computation to see what direction the stepper is going and if the wrong direction they swap because X & Y always seem to head in the correct direction is it possible to get Z axis to do the same ?? the bed moves in the wrong direction for between 10 & 30mm (not measured but a guess) before it reverses direction and heads towards the correct endstop. Or is this a different bug that needs a new issue raised or have I not searched hard enough ? again Thank You @ellensp for this fix.

cp2004, 2021-09-08:
There's over 500 bug reports just like this and the only people that will fix them are someone who dedicates their spare time to help out others. You have to accept that not everything will get done because people contributing have personal interest in issues that affect them and will tend towards fixing those, not some random other boring bug report. If people were paid, it's a different story, but they're not.

As for the later issue, it sounds like a configuration issue, just need to change the homing direction. If you want help with configuration get in contact on the Marlin Discord or other forum on the internet, there's lots of other knowledgeable people about there rather than here.

CBDesignS, 2021-09-08:
@cp2004 I have tried other places but when one person says one thing then the next says exactly the opposite what do you do ??? 

p.s it is only the initial power up that the Z axis goes wrong for 10-30mm then swaps direction. every other home works correctly after the initial powerup so changing configuration direction is counter intuitive that is unless Marlin once powered up knows that the direction is wrong so it just fixes it on the fly keeping the user in the dark . If this is how it works then a suggestion would have been to leave the direction wrong so the user knew there was a problem and it needed fixing instead of fixing it on quiet letting the user think they had the configuration nailed down.. BTW Once I do get this sorted I will donate to the funds as a self build of Marlin is far better than the out of date crap & support from both BTT & MKS that seem only interested in dumping new hardware out than supporting their own older hardware.  Thanks again Charlie

ellensp, 2021-09-09:
According to main developer you should enable #define HOME_Z_FIRST

CBDesignS, 2021-09-09:
> #define HOME_Z_FIRST

Thank you for prompt reply @ellensp, this is only of use if no probe is used. The machine I have problems with is an Ender 5 "upgraded" with a 3d probe fitted so it needs to use Z_SAFE_HOMING so the probe can find the center of the bed and it can complete the z home correctly.  

again thankyou for fixing the initial bed moving bug that has now highlighted another problem I now need to resolve or just ignore and live with because I have no idea if it is a feature or prob a bug as it only happens once at power up/ hard reset of the system.