MarlinFirmware/Marlin #21854: [BUG] pobe.cpp fails to compile with `DELTA`, a probe with quiet probing and `PROBING_STEPPERS_OFF`

sgparry, 2021-05-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Since March 2nd commit (https://github.com/MarlinFirmware/Marlin/commit/35791c6371dad924f9a3f0fb3aebacfd298ecc35#diff-844640fa2c3026003779111f0d4594b9370678140e782a408409d57a9c1dc809), platformio build with `DELTA`, a bed probe that results in `HAS_QUIET_PROBING` and with `PROBING_STEPPERS_OFF` defined fails thus:
```
Marlin/src/module/probe.cpp: In static member function 'static void Probe::set_probing_paused(bool)':
Compiling .pio/build/mega2560/lib1ed/U8glib-HAL/clib/u8g_dev_flipdisc_2x7.c.o
Marlin/src/module/probe.cpp:265:24: error: 'old_trusted' was not declared in this scope
         axis_trusted = old_trusted;
                        ^~~~~~~~~~~
Marlin/src/module/probe.cpp:265:24: note: suggested alternative: 'axis_trusted'
         axis_trusted = old_trusted;
                        ^~~~~~~~~~~
                        axis_trusted
Compiling .pio/build/mega2560/lib1ed/U8glib-HAL/clib/u8g_dev_gprof.c.o
Compiling .pio/build/mega2560/lib1ed/U8glib-HAL/clib/u8g_dev_ht1632.c.o
*** [.pio/build/mega2560/src/src/module/probe.cpp.o] Error 1
```
scrutiny of the source code shows that if `DELTA` is defined, then `old_trusted` is not declared, but is still referenced.

```
#if HAS_QUIET_PROBING

  #ifndef DELAY_BEFORE_PROBING
    #define DELAY_BEFORE_PROBING 25
  #endif

  void Probe::set_probing_paused(const bool dopause) {
    TERN_(PROBING_HEATERS_OFF, thermalManager.pause(dopause));
    TERN_(PROBING_FANS_OFF, thermalManager.set_fans_paused(dopause));
    #if ENABLED(PROBING_STEPPERS_OFF)   
      IF_DISABLED(DELTA, static uint8_t old_trusted);
      if (dopause) {
        #if DISABLED(DELTA)
          old_trusted = axis_trusted;
          DISABLE_AXIS_X();
          DISABLE_AXIS_Y();
        #endif
        disable_e_steppers();
      }
      else {
        #if DISABLED(DELTA)
          if (TEST(old_trusted, X_AXIS)) ENABLE_AXIS_X();
          if (TEST(old_trusted, Y_AXIS)) ENABLE_AXIS_Y();
        #endif
        axis_trusted = old_trusted;
      }
    #endif
    if (dopause) safe_delay(_MAX(DELAY_BEFORE_PROBING, 25));
  }

#endif // HAS_QUIET_PROBING
```
I would propose a patch, but I have not a clue what would fix this safely. I can hear my hotend crashing into my print bed even thinking about it.

### Bug Timeline

March 2nd 2021 https://github.com/MarlinFirmware/Marlin/commit/35791c6371dad924f9a3f0fb3aebacfd298ecc35#diff-844640fa2c3026003779111f0d4594b9370678140e782a408409d57a9c1dc809

### Expected behavior

I expect Marlin to compile.

### Actual behavior

```
Marlin/src/module/probe.cpp: In static member function 'static void Probe::set_probing_paused(bool)':
Compiling .pio/build/mega2560/lib1ed/U8glib-HAL/clib/u8g_dev_flipdisc_2x7.c.o
Marlin/src/module/probe.cpp:265:24: error: 'old_trusted' was not declared in this scope
         axis_trusted = old_trusted;
                        ^~~~~~~~~~~
Marlin/src/module/probe.cpp:265:24: note: suggested alternative: 'axis_trusted'
         axis_trusted = old_trusted;
                        ^~~~~~~~~~~
                        axis_trusted
Compiling .pio/build/mega2560/lib1ed/U8glib-HAL/clib/u8g_dev_gprof.c.o
Compiling .pio/build/mega2560/lib1ed/U8glib-HAL/clib/u8g_dev_ht1632.c.o
*** [.pio/build/mega2560/src/src/module/probe.cpp.o] Error 1
```

### Steps to Reproduce

- Download and install PlatformIO as per instructions
- clone the 2.0.x source code
- set up a delta based configuration that includes the following defines:
```
#define DELTA
#define FIX_MOUNTED_PROBE
#define PROBING_STEPPERS_OFF
```
- attempt compile

### Version of Marlin Firmware

2.0.8

### Printer model

Velleman Vertex Delta K8800

### Electronics

Stock

### Add-ons

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6447541/Configuration.zip)

sjasonsmith, 2021-08-17:
I posted a change which I believe will resolve this, although I haven't actually tested it.

I think that you could have gotten around this by enabling PROBING_ESTEPPERS_OFF instead of PROBING_STEPPERS_OFF. On a Delta those would be effectively the same.

But...I don't see why PROBING_STEPPERS_OFF should preclude Deltas, and I think my new code is simpler anyway.

thisiskeithb, 2021-09-15:
Resolved in https://github.com/MarlinFirmware/Marlin/pull/22581

closed 2021-09-15 by thisiskeithb