MarlinFirmware/Marlin #21857: [BUG] M605 in duplication and mirror mode does not extrude both extruders if DISABLE_INACTIVE_EXTRUDER is defined

CyberDeck, 2021-05-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have an IDEX printer with DUAL_X_CARRIAGE set in Configuration_adv.h.
In the Configuration.h is enabled (defined) the DISABLE_INACTIVE_EXTRUDER setting.

If I use both extruders in "full control" or "auto-park" mode I can extrude both extruders E0 and E1.

However, if I enable with "M605 S2 X100" the duplication or with "M605 S3 X100" the mirror mode only the left (i.e. E0) extruder is extruding. The other one is disabled.

### Bug Timeline

Dunno. Probably a new bug.

### Expected behavior

In "duplication" or "mirror mode" I expect that both hotends extrude and not only one of them.

### Actual behavior

Only the left (i.e. E0) extrudes. The other one does not extrude.

### Steps to Reproduce

1. Startup printer
2. Send following commands:
`G28`
`M605 S2 X100`
`G28 X`
`G0 X100 Y0 Z5`
`M302 P1 ; allow cold extrudes for test purpose`
`G1 E5 F50 ; only E0 extrudes and E1 errorneously stand still`

### Version of Marlin Firmware

Marlin 2.0.8 (May  9 2021 21:49:50) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0

### Printer model

EasyFrame3D (a modified ANET A8 with a steel frame and a custom IDEX configuration)

### Electronics

BTT SKR PRO v1.1

DrumClock, 2021-05-10:
**Ano, potvrzuji chybu.**
než to někdo opraví ...... nepoužívej funkci `DISABLE_INACTIVE_EXTRUDER` pokud chceš 
tiskonou v  `duplication` nebo `mirror` modu.

CyberDeck, 2021-05-10:
Yes, if I do not use the DISABLE_INACTIVE_EXTRUDER function everything works as expected.

mikejr83, 2021-05-12:
I can confirm that with a MKS Robin Pro and BTT GTR 1.0 this issue exists. It was not present in 2.0.7.2.

BonusNachos, 2021-05-12:
Another confirmation on this issue for the gtr

JesusvtR, 2021-05-20:
Same problem with SKR PRO V1.2

FancyChancie, 2021-05-22:
I can also confirm this issue (using Formbot/Vivedino T-REX 3).

DenisSemenov1970, 2021-06-03:
This problem appeared after the changes #20130

closed 2021-08-12 by github-actions[bot]