MarlinFirmware/Marlin #21861: [BUG] Menu navigation - wrong beahviour

davidtgbe, 2021-05-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Wrong behaviour navigating through menu.

https://streamable.com/aezxag

### Expected behavior

Going to configuration menu and back to menu should take you back to the status screen

### Actual behavior

Going to configuration menu and, quickly touching back icon, takes you to the "Z move" menu instead to the status screen

### Steps to Reproduce

1. Press configuration icon on status screen
2. Quickly, press back icon

### Version of Marlin Firmware

Latest bugfix

### Printer model

ET4

### Electronics

STM32F407 + TOUCH TFT 2.8 FSMC

ellensp, 2021-05-10:
Do you have DOUBLECLICK_FOR_Z_BABYSTEPPING enabled?

davidtgbe, 2021-05-10:
Yes. It is enabled.

ellensp, 2021-05-10:
And if you disable DOUBLECLICK_FOR_Z_BABYSTEPPING ?

davidtgbe, 2021-05-10:
With DOUBLECLICK_FOR_Z_BABYSTEPPING works fine.

rhapsodyv, 2021-05-12:
Yes, it's the intended behaviour. Currently, color ui doesn't handle double touch. So, this is the way the dev created to mimic the 12864 LCD with encoder. In fact, old lcd and color ui have the very same behaviour.

To mitigate it, we can have a config to increase/decrease the interval? Or add full support for double click on color ui... Anyway, this is not a bug, but a feature! :-)

closed 2021-05-12 by rhapsodyv

davidtgbe, 2021-05-12:
Understood.
My first impression was that it was a bug. It is not a very intuitive behaviour IMHO. It wasn't until I read the comments in code that I realised it was the DOUBLECLICK_FOR_Z_BABYSTEPPING default behaviour I had been looking for so long  xD.