MarlinFirmware/Marlin #21862: [BUG] UBL extrapolation leads to insane z hop movements for prints outside probing area

Phil1988, 2021-05-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am using UBL on the latest firmware Marlin-2.0.7.2

When I print something outside of the probed area, my z-axis is doing some strange movements.
Its like z-hop on retraction, but its no retraction ;)

It only appears on the right or on the back of the probed area.
Here you can see my mesh:
[image]

I guess it has something to do with the extrapolation of the points that cant be reached.

My bed size is 302x307mm.

The mesh probing reports this back:
Send: G29 T
Recv: Bed Topography Report:
Recv: 
Recv:     ( 20,287)                                                                                                (283,287)
Recv:           0          1           2           3          4          5          6          7          8          9
Recv:  9 | -0.942  -0.929  -0.968  -1.024  -1.056  -1.027  -1.106  -1.070  -1.059  -1.071
Recv:     |
Recv:  8 | -1.117  -1.078  -1.122  -1.099  -1.060  -1.064  -1.058  -1.087  -1.011  -0.960
Recv:     |
Recv:  7 | -0.993  -0.955  -1.026  -1.010  -1.052  -1.018  -1.107  -1.099  -1.048  -1.042
Recv:     |
Recv:  6 | -1.157  -1.086  -1.165  -1.097  -1.093  -1.085  -1.086  -1.113  -0.999  -1.001
Recv:     |
Recv:  5 | -1.187  -1.136  -1.100  -1.084  -1.118  -1.082  -1.110  -1.074  -1.041  -1.072
Recv:     |
Recv:  4 | -1.203  -1.172  -1.141  -1.188  -1.169  -1.138  -1.152  -1.111  -1.088  -1.129
Recv:     |
Recv:  3 | -1.235  -1.179  -1.160  -1.129  -1.168  -1.110  -1.106  -1.050  -1.012  -1.073
Recv:     |
Recv:  2 | -1.162  -1.056  -1.075  -1.079  -1.125  -1.097  -1.103  -1.082  -1.071  -1.087
Recv:     |
Recv:  1 | -1.131  -1.055  -1.031  -1.013  -1.052  -0.986  -0.975  -0.944  -0.905  -0.949
Recv:     |
Recv:  0 | -0.990  -0.934  -0.965  -0.942  -0.986  -0.962  -0.974  -0.955  -1.029  -1.061
Recv:           0          1           2          3           4          5           6          7          8         9
Recv:     ( 20, 20)                                                                                                 (283, 20)

So its about 20mm that can be probed.

Here I am printing a 290x290mm part, which shows the problem:
https://www.youtube.com/watch?v=WxJ-mZ6OyK4

Tested with Cura, Slic3r, PrusaSlicer and SuperSlicer -> its not Slicer related.

Tested without UBL. Works without jumps -> its UBL related.

### Bug Timeline

I dont think this is new.

### Expected behavior

UBL should level the bed in a way, that a smooth movement in z direction without "z jumps" is possible.

### Steps to Reproduce

1. homing the bed with: G28
2. probing with: G29 P
3. save the mesh with: G29 S1
4. set fade heigt: G29 F3.0 
5. activate UBL: G29 A
6. Start a print with G28, G29 L1, G29 J  -> z jumps

### Version of Marlin Firmware

Marlin-2.0.7.2

### Printer model

Creality CR10 

### Electronics

SKR 1.3, BLtouch, dual z, magnetic printbed

### Host Software

OctoPrint

### Additional information & file uploads

My Configuration.h and Configuration_adv.h:

[Marlin Config.zip](https://github.com/MarlinFirmware/Marlin/files/6451286/Marlin.Config.zip)

XDA-Bam, 2021-05-12:
You say you run `G29 P` before the print, but I always thought you need to specify which P you want UBL to run (1-6). The [documentation](https://marlinfw.org/docs/gcode/G029-ubl.html) also does not list `G29 P` as a valid command. Is this a typo or am I missing something?

In any case, did you try `G29 P1` followed by `G29 P3` to autofill the missing mesh points?

Phil1988, 2021-05-13:
Oh hey @XDA-Bam : 
Yes it was a typo. 
I am using `G29 P1` once for measuring the bed and `G29 J` at the start of the gcode of every print.

I corrected the "P1" at my post above.

I did not G29 P3.
The reson is, that I used `#define MESH_INSET 20`.
So the probe was capable to move to every point inside that "inset".

But I also changed it down to `#define MESH_INSET 8`.
The problem is still existent, but the "dangerous zone" on the right and back, where these z jumps happens, shrinked.

So I guess it might have something to do with that mesh inset?

Just for reference, I give you the new bed data, with that 8mm inset.
The bed offsets now looks pretty different from above, because this time I did not G29 P1 it with 60°C (PLA) but with 115°C (ABS):

```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  8,299)                                                              (295,299)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 | +0.142  +0.155  +0.150  +0.100  +0.048  +0.060  +0.132  +0.095  +0.097  +0.270
Recv:    |
Recv:  8 | -0.035  +0.045  +0.042  -0.013  -0.030  +0.053  +0.085  +0.122  +0.102  +0.280
Recv:    |
Recv:  7 | +0.032  +0.117  +0.048  +0.060  +0.037  +0.120  +0.085  +0.112  +0.137  +0.242
Recv:    |
Recv:  6 | -0.060  +0.012  -0.015  +0.017  +0.040  +0.037  +0.017  +0.085  +0.087  +0.257
Recv:    |
Recv:  5 | -0.135  +0.007  -0.050  -0.023  -0.025  -0.030  +0.022  +0.040  +0.007  +0.202
Recv:    |
Recv:  4 | -0.150  -0.062  -0.095  -0.070  -0.105  -0.040  -0.013  +0.017  +0.005  +0.120
Recv:    |
Recv:  3 | -0.150  -0.060  -0.023  +0.012  -0.053  -0.003  +0.022  +0.045  +0.035  +0.178
Recv:    |
Recv:  2 | -0.133  -0.018  -0.008  +0.017  -0.015  +0.007  +0.032  +0.053  +0.055  +0.160
Recv:    |
Recv:  1 | -0.088  +0.053  +0.065  +0.122  +0.075  +0.130  +0.150  +0.175  +0.185  +0.250
Recv:    |
Recv:  0 |[+0.077] +0.145  +0.132  +0.175  +0.082  +0.145  +0.100  +0.060  +0.005  +0.045
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     (  8,  8) 
```

Phil1988, 2021-05-13:
I tested it the last couple of hours and here are my findings:

This time I didnt use the mesh inset. 
The probe does not measure the outer most points on the edges.
I tested this with `#define PROBING_MARGIN 5`.
This leads to incorrect calculated (smart fill) points on the edges.
In my case it would be better to just use a weighted average of the nearby points.

```
Here is my mesh mit 15x15 points:
Recv:     (  0,307)                                                                                                      (303,307)
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv: 14 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 13 |   .     +0.070  +0.107  +0.107  +0.048  +0.040  +0.017  -0.020   0.000  +0.005  +0.017  -0.008  +0.030  +0.020    .
Recv:    |
Recv: 12 |   .     -0.055  -0.030  +0.017  -0.080  -0.005  -0.045  -0.015  -0.005  -0.030  +0.035  +0.010  +0.065  +0.097    .
Recv:    |
Recv: 11 |   .     -0.030  +0.022  -0.010  +0.007  -0.035  -0.013  +0.020  +0.065  -0.020  -0.005  +0.017  +0.037  +0.062    .
Recv:    |
Recv: 10 |   .     -0.005  +0.030  +0.022  +0.058  -0.013  +0.022   0.000  +0.022  -0.003  -0.040  -0.010  +0.027  +0.020    .
Recv:    |
Recv:  9 |   .     -0.187  -0.118  -0.143  -0.125  -0.058  -0.038  -0.053  -0.023  +0.020  +0.005  +0.025  +0.050  +0.048    .
Recv:    |
Recv:  8 |   .     -0.168  -0.108  -0.123  -0.067  -0.050  -0.015  -0.075  -0.050  -0.043  -0.020  -0.005  +0.035  +0.007    .
Recv:    |
Recv:  7 |   .     -0.158  -0.118  -0.113  -0.067  -0.085  -0.103  -0.123  -0.070  -0.067  -0.098  -0.105  -0.120  -0.065    .
Recv:    |
Recv:  6 |   .     -0.312  -0.223  -0.200  -0.215  -0.197  -0.148  -0.138  -0.103  -0.108  -0.128  -0.065  -0.030  -0.045    .
Recv:    |
Recv:  5 |   .     -0.228  -0.187  -0.150  -0.173  -0.140  -0.140  -0.143  -0.095  -0.095  -0.045  -0.035  -0.025  -0.010    .
Recv:    |
Recv:  4 |   .     -0.170  -0.143  -0.105  -0.105  -0.105  -0.093  -0.123  -0.115  -0.108  -0.085  -0.090  -0.077  -0.123    .
Recv:    |
Recv:  3 |   .     -0.223  -0.158  -0.115  -0.115  -0.105  -0.060  -0.090  -0.062  -0.050  -0.020  +0.002  +0.010  -0.003    .
Recv:    |
Recv:  2 |   .     -0.168  -0.090  -0.013  -0.058  -0.060  -0.020  -0.050  -0.005  +0.048  +0.035  +0.070  +0.060  +0.030    .
Recv:    |
Recv:  1 |   .     -0.062  -0.025  +0.007  -0.023  -0.008  -0.023  -0.035   0.000  -0.013  -0.028  -0.055  -0.090  -0.135    .
Recv:    |
Recv:  0 |[  .   ]   .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv:     (  0,  0)    
```

Filled with `G29 P3`:
```
Recv:     (  0,307)                                                                                                      (303,307)
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv: 14 | +0.195  +0.195  +0.245  +0.197  +0.175  +0.085  +0.080  -0.020  +0.005  +0.040  +0.017  -0.008  +0.030  +0.020  +0.020
Recv:    |
Recv: 13 | +0.070  +0.070  +0.107  +0.107  +0.048  +0.040  +0.017  -0.020   0.000  +0.005  +0.017  -0.008  +0.030  +0.020  +0.020
Recv:    |
Recv: 12 | -0.055  -0.055  -0.030  +0.017  -0.080  -0.005  -0.045  -0.015  -0.005  -0.030  +0.035  +0.010  +0.065  +0.097  +0.130
Recv:    |
Recv: 11 | -0.030  -0.030  +0.022  -0.010  +0.007  -0.035  -0.013  +0.020  +0.065  -0.020  -0.005  +0.017  +0.037  +0.062  +0.088
Recv:    |
Recv: 10 | -0.005  -0.005  +0.030  +0.022  +0.058  -0.013  +0.022   0.000  +0.022  -0.003  -0.040  -0.010  +0.027  +0.020  +0.020
Recv:    |
Recv:  9 | -0.187  -0.187  -0.118  -0.143  -0.125  -0.058  -0.038  -0.053  -0.023  +0.020  +0.005  +0.025  +0.050  +0.048  +0.048
Recv:    |
Recv:  8 | -0.168  -0.168  -0.108  -0.123  -0.067  -0.050  -0.015  -0.075  -0.050  -0.043  -0.020  -0.005  +0.035  +0.007  +0.007
Recv:    |
Recv:  7 | -0.158  -0.158  -0.118  -0.113  -0.067  -0.085  -0.103  -0.123  -0.070  -0.067  -0.098  -0.105  -0.120  -0.065  -0.010
Recv:    |
Recv:  6 | -0.312  -0.312  -0.223  -0.200  -0.215  -0.197  -0.148  -0.138  -0.103  -0.108  -0.128  -0.065  -0.030  -0.045  -0.045
Recv:    |
Recv:  5 | -0.228  -0.228  -0.187  -0.150  -0.173  -0.140  -0.140  -0.143  -0.095  -0.095  -0.045  -0.035  -0.025  -0.010  +0.005
Recv:    |
Recv:  4 | -0.170  -0.170  -0.143  -0.105  -0.105  -0.105  -0.093  -0.123  -0.115  -0.108  -0.085  -0.090  -0.077  -0.123  -0.123
Recv:    |
Recv:  3 | -0.223  -0.223  -0.158  -0.115  -0.115  -0.105  -0.060  -0.090  -0.062  -0.050  -0.020  +0.002  +0.010  -0.003  -0.003
Recv:    |
Recv:  2 | -0.168  -0.168  -0.090  -0.013  -0.058  -0.060  -0.020  -0.050  -0.005  +0.048  +0.035  +0.070  +0.060  +0.030  +0.030
Recv:    |
Recv:  1 | -0.062  -0.062  -0.025  +0.007  -0.023  -0.008  -0.023  -0.035   0.000  -0.013  -0.028  -0.055  -0.090  -0.135  -0.135
Recv:    |
Recv:  0 |[+0.045] +0.043  +0.040  +0.027  +0.012  +0.045  -0.023  -0.020  +0.005  -0.013  -0.028  -0.055  -0.090  -0.135  -0.135
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv:     (  0,  0)                                  
[... cut ...]
If you have a look at the top left side, you can see, that smart fill adds ~0.15mm to the edge points... way too much.
Is there a way to use a weighted average or even the same height as the nearest point(s)?
I know I can use my LCD and measure/fill each point manually, but doing this for 60 points will need 2hours :D
So if there is a handy possibility to set those points manually in a ralativley fast way? (I assume, that my assessments for each point would be better)

The good news: doing it without mesh inset make the initial error/bug dissapear

XDA-Bam, 2021-05-13:
Looking at the interpolated points, everything looks good and as documented for `G29 P3`. That is: nearest neighbour for downward curved regions, linear interpolation for upward curved regions. I agree that his interpolation method happens to produce results which don't look ideal for your bed . Sadly, there is currently no implementation in UBL to influence the method of interpolation - you can either go with the default, or use `G29 P3 C#` to set all outside points to a constant value `#`. Certainly not ideal.

That being said, I still don't understand why the printer Z-hops that much when using a non-interpolated mesh and printing in the out-of-mesh region. It may be some safety feature, because the bed height is unknown in those regions and UBL wants to avoid leaving long scratches. I'm not familiar enough with the code to tell.

Different point: After looking at it again, your probing results look a bit noisy/inconsistent. I noticed that you use `#define BLTOUCH_DELAY 60`. Anything below 200 ms throws a compilation warning for me. And there has to be a reason for that. I recommend you test something around `#define BLTOUCH_DELAY 250` to see if your mesh looks any different / smoother. You might also want to try `#define MULTIPLE_PROBING 3`, to test if that improves your mesh. A smoother mesh would most likely also improve the results for the points interpolated by `G29 P3`. But it may very well be that your bed actually looks like the mesh tells us.

Phil1988, 2021-05-14:
@XDA-Bam 
The `#define BLTOUCH_DELAY 60` was a test, because my probing is extremely slow.

to make this possible I changed the the Marlin\src\inc\SanityCheck.h like this:
`#if BLTOUCH_DELAY < 50`
It did not get much better with this.
I guess that my wireing from the BLtouch to the SKR 1.3 is too long (about 150cm) and results into some interferences or something...

You can see the "speed" here:
https://youtu.be/M3M_IrbyKkk

The mesh is correct and wont be smoother if I do longer delays or multiple (+ extra) probes.
The BLtouch itslf works accurate but slow.
I tested it once with M48 P20 X150 Y150 V2 ;and was really happy with the results.

About the z-jumps, you might be right with the "safety feature", but that can only tell us someone who is more involved.

closed 2021-07-24 by github-actions[bot]