MarlinFirmware/Marlin #21866: Emergency Stop but not M112

Maazshakir, 2021-05-10:
### Is your feature request related to a problem? Please describe.

NO

### Are you looking for hardware support?

NO

### Describe the feature you want

Emergency stop gcode but not M112 because after sending M112 printer is halted , I would like an emergency stop gcode which wont halt the printer but pnly disable motors , heating and stuff like that and after that we can normally resume our work. this will be useful while calibrating your printer for example if for some reason bltouch doesnt work we can stop the printer working immdiately

yet-another-average-joe, 2021-05-19:
M0 with additional parameters for motors and heating would be great.

slowbro, 2021-06-15:
I think the `FREEZE_FEATURE` solves this: https://github.com/MarlinFirmware/Marlin/pull/17462

It was added May 13.

Maazshakir, 2021-06-15:
Ok with that I am closing this issue

closed 2021-06-15 by Maazshakir