MarlinFirmware/Marlin #21873: [FR] Support for I2C Support on alternate Pins on LPC1768

Schwuuuuup, 2021-05-11:
### Is your feature request related to a problem? Please describe.

The LPC1768 has 3 I2C Ports, `I2C.0`, `I2C.1` and `I2C.2`, these can be used for attaching displays.

The default is `I2C.1`, this is set here:
https://github.com/MarlinFirmware/Marlin/blob/cb4e3889ece8fd07030f4c2eeaa781679cd8414e/Marlin/src/HAL/LPC1768/include/i2c_util.h#L30-L42
and it is evaluated here:
https://github.com/MarlinFirmware/Marlin/blob/cb4e3889ece8fd07030f4c2eeaa781679cd8414e/Marlin/src/HAL/LPC1768/include/i2c_util.c#L41-L64

Via this all three I2C busses can be chosen, but I2C.1 has an additional feature: it can be put on pins `P0.0`/`P0.1` and on P0.19/P0.20. This fourth option can not be chosen right now.

### Describe the feature you want

The BigTreTech SKR 1.3 Uses all I2C-Pins of the three available options as GPIO for other purposes. If one would be able to chose "`I2C.1 on P0.19/20`" these pins could be accessed via the EXP1-Header of the SKR 1.3 opening up the possibillity to use any supported I2C Display, like 128x64 SSD1306 OLEDs

This could be achieved by 
setting bits 0-3 of `PINSEL0 `to 0 (disabling `I2C.1` on pins P0.0/1) and
setting bits 6-9 of `PINSEL1` to 1 (enabling `I2C.1` on pins P0.19/20)

Unfortunately I'm a noob regarding C and I'm to unsure how `PINSEL_ConfigPin()` is implemented (is it a macro?) otherwise I would send in a pull request.

### Additional context

Any answer would be appreciated, even if it's a "won't fix"

Best regards
TOM

descipher, 2021-05-12:
This is not a feature function, you need to map your pins in a custom way. 
The code you are looking at is not implemented based on pin selections, I2C functions are done with the arduino libs and wire.h
For the BBT SKR 1.3 see src\pins\lpc1768\pins_BTT_SKR_V1_3.h and pins_BTT_SKR_common.h
You would need to comment out any defines for E1 and then assign the aux's as desired.
This FR can be closed.

Schwuuuuup, 2021-05-12:
I'm not sure this answer fits the question.
This is about the u8glib drivers for displays. they use the I2C bus configured by `I2C_MASTER_ID`in `configuration_adv.h` . If I'm not mistaken this is not directly related to the pin definitions in pins_BTT_SKR_V1_3.h and pins_BTT_SKR_common.h 

Yes you define what pin is mapped to what function there, but first you have to configure what pins are availeble. This is done in  `Marlin/Marlin/src/HAL/LPC1768/include/i2c_util.h` 

So in order to create a custom mapping you have to be able to put I2C.1 on different pins

descipher, 2021-05-12:
You can only use P0_00 & P0_01 if you do not have a second extruder .. e.g. set E1 pins to -1
Trying to switch I2C to P0_19 & 20 will not work. 
#define Y_STEP_PIN                         P0_19
#define Y_DIR_PIN                          P0_20

e.g.
 https://github.com/MarlinFirmware/Marlin/issues/13257

Schwuuuuup, 2021-05-13:
Oh... There is my mistake: I confused `P1_19/20` wich are present on the `EXP1` connector with `P0_19/20` which are used on the Y-axis, D'oh!

closed 2021-05-13 by Schwuuuuup