MarlinFirmware/Marlin #21878: Missing "ok" on M190 "echo:busy: processing"

Japhei, 2021-05-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

### Description

The print bed heats up to the requested 60 °, but does not return "ok"! Instead, I get the temperatures a few times and an "echo: busy: processing", that's it. More details can be found in the console log. I hope the prefix isn't too confusing.

### Console log:
The initialisation (M115):
```
[SerialConnection] sending data: M115
[SerialInputListener] {ttyACM0} FIRMWARE_NAME:Marlin bugfix-2.0.x (May 12 2021 16:07:34) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
[SerialInputListener] {ttyACM0} Cap:SERIAL_XON_XOFF:0
[SerialInputListener] {ttyACM0} Cap:BINARY_FILE_TRANSFER:0
[SerialInputListener] {ttyACM0} Cap:EEPROM:0
[SerialInputListener] {ttyACM0} Cap:VOLUMETRIC:1
[SerialInputListener] {ttyACM0} Cap:AUTOREPORT_TEMP:1
[SerialInputListener] {ttyACM0} Cap:PROGRESS:0
[SerialInputListener] {ttyACM0} Cap:PRINT_JOB:1
[SerialInputListener] {ttyACM0} Cap:AUTOLEVEL:0
[SerialInputListener] {ttyACM0} Cap:RUNOUT:0
[SerialInputListener] {ttyACM0} Cap:Z_PROBE:0
[SerialInputListener] {ttyACM0} Cap:LEVELING_DATA:0
[SerialInputListener] {ttyACM0} Cap:BUILD_PERCENT:0
[SerialInputListener] {ttyACM0} Cap:SOFTWARE_POWER:0
[SerialInputListener] {ttyACM0} Cap:TOGGLE_LIGHTS:0
[SerialInputListener] {ttyACM0} Cap:CASE_LIGHT_BRIGHTNESS:0
[SerialInputListener] {ttyACM0} Cap:EMERGENCY_PARSER:0
[SerialInputListener] {ttyACM0} Cap:PROMPT_SUPPORT:0
[SerialInputListener] {ttyACM0} Cap:SDCARD:0
[SerialInputListener] {ttyACM0} Cap:REPEAT:0
[SerialInputListener] {ttyACM0} Cap:SD_WRITE:0
[SerialInputListener] {ttyACM0} Cap:AUTOREPORT_SD_STATUS:0
[SerialInputListener] {ttyACM0} Cap:LONG_FILENAME:0
[SerialInputListener] {ttyACM0} Cap:THERMAL_PROTECTION:1
[SerialInputListener] {ttyACM0} Cap:MOTION_MODES:0
[SerialInputListener] {ttyACM0} Cap:ARCS:1
[SerialInputListener] {ttyACM0} Cap:BABYSTEPPING:0
[SerialInputListener] {ttyACM0} Cap:CHAMBER_TEMPERATURE:0
[SerialInputListener] {ttyACM0} Cap:COOLER_TEMPERATURE:0
[SerialInputListener] {ttyACM0} Cap:MEATPACK:0
[SerialInputListener] {ttyACM0} ok
```
Print start (M140, M105, M190):
```
[SerialConnection] sending data: M140 S60
[SerialInputListener] {ttyACM0} ok
[SerialConnection] sending data: M105
[SerialInputListener] {ttyACM0} ok T:20.86 /0.00 B:19.38 /60.00 @:0 B@:0
[SerialConnection] sending data: M190 S60
[SerialInputListener] {ttyACM0} T:20.86 /0.00 B:19.38 /60.00 @:0 B@:0 W:?
[SerialInputListener] {ttyACM0} T:20.75 /0.00 B:19.38 /60.00 @:0 B@:0 W:?
[SerialInputListener] {ttyACM0} echo:busy: processing
[DefaultGcode] Can't understand this type of response: (echo:busy: processing) ignoring it... 
```

### Bug Timeline

new bug

### Expected behavior

Heating up to 60 degree and sending "ok"

### Actual behavior

Heating up and sending "echo:busy: processing"

### Steps to Reproduce

Start any print with heatbed (M190).

### Version of Marlin Firmware

Marlin bugfix-2.0.x (May 12 2021 16:07:34) 

### Printer model

custom

### Electronics

SKR_1.4_TURBO

### Add-ons

BTT_MOTOR_EXPANSION

### Your Slicer

Cura

### Host Software

Other (explain below)

### Additional information & file uploads

I use my own Host Software, but I don't think there is a problem with it. It runs on older Marlin versions without any problems.

If you need any other recources, just let me know.

Otherwise I wish you happy debugging! ;D

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/6466985/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/6466988/Configuration_adv.txt)
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/6466997/platformio.txt)

**Note:** Enable `#define BTT_MOTOR_EXPANSION 1` in `Marlin\src\pins\lpc1768\pins_BTT_SKR_common.h`

Japhei, 2021-05-12:
No there is  only one `echo:busy: processing` and nothing else. The bed heats up to the requested temperature but there is no `ok`! (Even after half an hour)
And no, the serial connection stays open. (At least from host side)
Like I said, the host software works fine with other printers (e.g. Anycubic i3 Mega) with older Marlin versions.

ellensp, 2021-05-13:
you can disable these "echo:busy: processing" messages with disabling HOST_KEEPALIVE_FEATURE

ellensp, 2021-05-13:
Just tested this. same hardware and your Configs. Cannot replicate

```
M190 S60
15:21:33.349 >  T:52.42 /0.00 B:24.38 /60.00 @:0 B@:0 W:?
15:21:34.349 >  T:52.48 /0.00 B:24.38 /60.00 @:0 B@:127 W:?
15:21:35.349 >  T:52.42 /0.00 B:24.22 /60.00 @:0 B@:127 W:?
15:21:35.353 > echo:busy: processing
15:21:36.349 >  T:52.42 /0.00 B:24.22 /60.00 @:0 B@:127 W:?
15:21:37.349 >  T:52.48 /0.00 B:24.38 /60.00 @:0 B@:127 W:?
15:21:37.353 > echo:busy: processing
...
15:22:18.349 >  T:52.35 /0.00 B:59.79 /60.00 @:0 B@:127 W:4
15:22:19.349 >  T:52.48 /0.00 B:59.84 /60.00 @:0 B@:127 W:3
15:22:19.354 > echo:busy: processing
15:22:20.349 >  T:52.48 /0.00 B:60.00 /60.00 @:0 B@:127 W:2
15:22:21.349 >  T:52.42 /0.00 B:60.00 /60.00 @:0 B@:127 W:1
15:22:21.354 > echo:busy: processing
15:22:22.349 >  T:52.48 /0.00 B:60.05 /60.00 @:0 B@:127 W:0
15:22:23.074 > ok

```

rhapsodyv, 2021-05-13:
Comparing logs, it seems that @Japhei log had all the communication halted, as it doesn’t even log any more temperature updates. So it’s not only the “ok” that is missing in his log….

ellensp, 2021-05-13:
The log provided just looks incomplete to me.
All 9 lines of it
@Japhei can you provide a longer log, something that shows it actually reaches temperate ( B:60.00 is temp reached, / 60.00 is temp set)

Japhei, 2021-05-13:
@ellensp there is no longer log, because the communication stops like @rhapsodyv said. I noticed that the bed is heated up because the LED isn’t lighting anymore and after restarting the temperature is right below 60.
I just tested to send the commands manually in Arduino-IDE Serial-monitor and under Linux screen. Surprisingly this works fine. So, I think the problem is host-software related and has nothing to do with Marlin. I’m investigating the issue and I’ll close this post as soon as I’m sure.

Japhei, 2021-05-13:
I was able to locate the problem in the host-software. It's not Marlin related.
Nevertheless thanks for your help.

closed 2021-05-13 by Japhei