MarlinFirmware/Marlin #21881: [BUG] (Probing Failed Anycubic Kossel Linear Plus)

pasikov123, 2021-05-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The z-probe sensor stopped working

### Bug Timeline

new bug

### Expected behavior

Autocalibrate

### Actual behavior

Stow z-probe
Probing Failed

### Steps to Reproduce

1. Configuration 
2. Delta Calibration 
3. Auto Calibration 
4. Deploy Z-probe
5. Stow z-probe

### Version of Marlin Firmware

Probing Failed

### Printer model

Anycubic Kossel Delta

### Electronics

Trigorilla/ btt skr 1.4

### Add-ons

marlin 2.0.8

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

#define ANYCUBIC_PROBE_VERSION 2

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6468055/Configuration.zip)

thisiskeithb, 2021-05-12:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ThePsyk8, 2021-05-12:
> ### Did you test the latest `bugfix-2.0.x` code?
> 
> Yes, and the problem still exists.
> 
> ### Bug Description
> 
> The z-probe sensor stopped working during level bed. Z home and m48 works.
It's strange. Help me please

> 
> ### Bug Timeline
> 
> new bug
> 
> ### Expected behavior
> 
> Autocalibrate
> 
> ### Actual behavior
> 
> Stow z-probe
> Probing Failed
> 
> ### Steps to Reproduce
> 
> 1. Configuration 
> 2. Delta Calibration 
> 3. Auto Calibration 
> 4. Deploy Z-probe
> 5. Stow z-probe
> 
> 
> ### Version of Marlin Firmware
> 
> Marlin 2.0.8/2.0.7.2/2.0.6.1
>
> ### Printer model
> 
> Longer lk4 (motherboard V08)
> 
> ### Electronics
> 
> Stm32f1
> 
> ### Add-ons
> 
> The sensor works on the First version 2.0 of Marlin.
> ### Your Slicer
> 
> Cura/Ideamaker
> 
> ### Host Software
> 
> 
> ### Additional information & file uploads

pasikov123, 2021-05-13:
Everything works on this firmware

thisiskeithb, 2021-05-13:
> Everything works on this firmware

Great!

closed 2021-05-13 by thisiskeithb