MarlinFirmware/Marlin #21886: [BUG]Printhead going out of boundaries randomly #21709 (re-occured)

doucettom, 2021-05-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

First print after bootup or a long pause when it prints, first trial is alway out of boundaries (like the steppers are doubling distance) so it crashes in X-axis belt-tensionner
Cancelling and reprinting the same .gcode file and works fine.
Printing from USB via Octoprint (0.5.3)

I first thought it was related to me slicing with Cura Legacy inside of Octoprint but recently move to Cura 4.8 and it still does that on at least two printers (same motherboard+stepper drivers (2209 UART), MKS SGEN_L V1 but different mecanics)

### Bug Timeline

2.0.6 up to Marlin bugfix-2.0.x (Apr 26 2021 13:18:52) for sure

### Expected behavior

I expected it to move to print start point (middle of the bed minus print object 1/2 more or less)

### Actual behavior

It looks it moves twice as far (and seemingly twice as fast as well) and crashes into X tensionner if I don't stop it beforehand.

Once it prints one, I can reprint the same file and it always works until the printer is rebooted.
Original issue:
https://github.com/MarlinFirmware/Marlin/issues/21709

### Steps to Reproduce

1-Boot printer
2-Upload sliced gcode to Octoprint
3-Start printing. Z axis raise a lot higher than it is supposed to and look like all movement are doubled.
4-Cancelling the print
5-Reprinting the same file works fine til a reboot
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6470594/Marlin.zip)

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Apr 26 2021)

### Printer model

Tronxy X3 and AdimLAB Gantry Pro

### Electronics

both with MKS SGEN_L v1 boards and MKS 2209 UART Steppers

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Gcodes.zip](https://github.com/MarlinFirmware/Marlin/files/6470602/Gcodes.zip)

XDA-Bam, 2021-05-14:
A couple of questions:
* Does it happen if you print from SD?
* Does it happen with all gcodes, or only specific ones?
* Are you using ArcWelder or other plugins, which affect the gcode transfer to the printer?

Your config looks fine to me. Two Z steppers, two extruders, MeatPack is enabled, S-curve enabled, all TMC parameters seem to be set correctly, otherwise it looks pretty standard.

ManuelMcLure, 2021-05-14:
Things I'd be concerned about is whether the motherboard has power via USB before the main power supply is turned on. In that case the TMC drivers would not be properly initialized and the micro steps could be wrong. Sending an `M122` (assuming `TMC_DEBUG` is enabled) will show what the drivers have set for micro steps.
If that's the case the solution is to either ensure that the motherboard isn't powered over USB, or to explicitly send a `M122 I` after the power supply is turned on.

The other possibility is that relative mode was enabled with `G91`. That can be resolved by explicitly adding `G90` to your start gcode.

doucettom, 2021-05-14:
> A couple of questions:
>
> * Does it happen if you print from SD?
> * Does it happen with all gcodes, or only specific ones?
> * Are you using ArcWelder or other plugins, which affect the gcode transfer to [...]

I never prints from SD Card
Happens with all gcode
Not using ArcWelder or anything

XDA-Bam, 2021-05-14:
> 
> I never prints from SD Card
> Happens with all gcode
> Not using ArcWelder or anything

The suggestions of @ManuelMcLure sound like useful things to check. If they don't solve it, please try printing from SD to exclude problems with the serial connection.

doucettom, 2021-05-14:
> > I never prints from SD Card
> > Happens with all gcode
> > Not using ArcWelder or anything
> 
> The suggestions of @ManuelMcLure sound like useful things to check. If they don't solve it, please try printing from SD to exclude problems with the serial connection.
@ManuelMcLure this makes a lot of sense. Can I insert M122 I in my start gcode ? Doing so would redo it every files not only after a reboot though. I'll need to double check the startup gcode as well.

ManuelMcLure, 2021-05-14:
It should be fine to add `M122 I` to your start gcode. I can't see any downside to that. I've always tried to avoid depending on existing state and ensuring that that I always explicitly initialize anything that needs initializing.

doucettom, 2021-05-14:
Send: M122
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok

ManuelMcLure, 2021-05-14:
That just means that once you apply power it can communicate. But Marlin doesn't know that you've applied full power so it doesn't reinitialize the TMC drivers correctly. Enable `TMC_DEBUG` and send `M122` again and I'll bet you see the wrong settings for micro steps.

doucettom, 2021-05-14:
I will. From my understanding, M122 I still works but no verbose w/o the TMC_DEBUG ?

ManuelMcLure, 2021-05-14:
Yes. But if you want to see the actual data from the TMC drivers (like micro steps and current settings) you need `TMC_DEBUG`.

closed 2021-07-24 by github-actions[bot]