MarlinFirmware/Marlin #21887: [BUG] 1st layer not printing level, despite apparently successful probing using BLTouch

Zjaaspoer, 2021-05-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Despite starting with G28 & G29 in my G-code file, and the firmware & BLTouch doing a proper bed levelling (the BLTouch seems to work as intended), the first layer comes out uneven. Please find some pictures below of a test print (this one). I hope the pictures show clearly that the nozzle is too close in the bottom left, and too far in the top right (top left is also a bit far, and bottom right a bit close).

I've done days of research, all over the web, Reddit, forums & YouTube and tried numerous fixes, to no avail

### Bug Timeline

Observed it since installing BLTouch

### Expected behavior

I expect the first layer to print out evenly around the build plate (consistent distance between the nozzle and the bed)

### Actual behavior

Inconsistent distance between nozzle and the bed. The inconsistency seems to 'slope' regularly along the bed: Distance is biggest top right and smallest bottom left, with gradual change in between

### Steps to Reproduce

1. make sure the gcode start code contains `G28` and then `G29` (and no subsequent `G28`)
2. slice https://www.thingiverse.com/thing:4077747 in Cura 4.9.0
3. start print (I used raspberry pi 4 with latest version of octoprint)
4. observe first (and only) layer

### Version of Marlin Firmware

bugfix-2.0.x branch -> 2.0.8

### Printer model

Ender 3

### Electronics

Creality 3D silent mainboard v4.2.7

### Add-ons

Creality 3D BL Touch auto bed levelling kit v1 & Creality glass bed

### Your Slicer

Cura

### Additional information & file uploads

**What I've done to try to fix / debug 1: Observe z compensation**

When I do a test print with a bed levelling at the start, I observe the z-axis go up and down during the print, suggesting the printer is trying to compensate based on the readings from the start of the print. It just seems it's not compensating enough (or too much).

When I run a `M420 V` I get (which implies it has the mesh loaded):

    Send: M420 V
    Communication timeout while idle, trying to trigger response from printer. Configure long running commands or     increase communication timeout if that happens regularly on specific commands or long moves.
    Recv: Bilinear Leveling Grid:
    Recv:       0      1      2      3      4
    Recv:  0 +1.245 +1.257 +1.282 +1.332 +1.342
    Recv:  1 +1.187 +1.167 +1.130 +1.127 +1.147
    Recv:  2 +1.082 +1.080 +1.057 +1.077 +1.085
    Recv:  3 +1.202 +1.147 +1.057 +1.000 +0.957
    Recv:  4 +1.192 +1.180 +1.117 +1.085 +1.027
    Recv:
    Recv: echo:Bed Leveling ON
    Recv: echo:Fade Height 10.00
    Recv: ok P15 B3

**What I've done to try to fix / debug 2: Level the bed as much as possible**

I've tried to level the bed as best as possible. As you can observe from the `M420 V` command the bed is pretty level. This was done using the Bed Level Visualizer plugin from OctoPrint.

**What I've done to try to fix / debug 3: I've changed from a 3x3 grid to a 5x5 grid**

As advised in several places the bed levelling is now done with a 5x5 grid. This didn't make a (noticeable) difference.

**What I've done to try to fix / debug 4: I've updated the firmware**

I used to run on the Crealtiy firmware. I've downloaded new firmware from `marlin.crc.id.au` (did that today, so using `Ender3-v4.2.7-BLTouch-20210511.bin`). Didn't help.

**What I've done to try to fix / debug 5: I've calibrated the z offset**

I've done a lot of tests, tweaking the z value to the current value, where part of the bed comes too close to the nozzle, and part of the bed stays too far away. So the z offset is not going to be able to improve anything I believe.

**What I've done to try to fix / debug 6: I've done all the regular hardware tweaks**

I've checked all the common things: Belts are tight, wheels are properly tightened, nothing is wobbly, z-axis is clean.

**What I've done to try to fix / debug 7: I've tried to add M420 S**

I've tried to add the `M420` command after the `G29` command (I know it shouldn't be needed, as `G29` enables bed levelling, but just wanted to make sure)

**Reference: My printer M503 settings**

    echo:  G21    ; Units in mm (mm)
    echo:  M149 C ; Units in Celsius
    echo:; Filament settings: Disabled
    echo:  M200 S0 D1.75
    echo:; Steps per unit:
    echo: M92 X80.00 Y80.00 Z400.00 E93.00
    echo:; Maximum feedrates (units/s):
    echo:  M203 X500.00 Y500.00 Z20.00 E50.00
    echo:; Maximum Acceleration (units/s2):
    echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
    echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
    echo:  M204 P500.00 R500.00 T500.00
    echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
    echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E15.00
    echo:; Home offset:
    echo:  M206 X0.00 Y0.00 Z0.00
    echo:; Auto Bed Leveling:
    echo:  M420 S1 Z10.00
    echo:  G29 W I0 J0 Z1.24499
    echo:  G29 W I1 J0 Z1.25749
    echo:  G29 W I2 J0 Z1.28249
    echo:  G29 W I3 J0 Z1.33249
    echo:  G29 W I4 J0 Z1.34249
    echo:  G29 W I0 J1 Z1.18749
    echo:  G29 W I1 J1 Z1.16749
    echo:  G29 W I2 J1 Z1.12999
    echo:  G29 W I3 J1 Z1.12749
    echo:  G29 W I4 J1 Z1.14749
    echo:  G29 W I0 J2 Z1.08249
    echo:  G29 W I1 J2 Z1.07999
    echo:  G29 W I2 J2 Z1.05749
    echo:  G29 W I3 J2 Z1.07749
    echo:  G29 W I4 J2 Z1.08499
    echo:  G29 W I0 J3 Z1.20249
    echo:  G29 W I1 J3 Z1.14749
    echo:  G29 W I2 J3 Z1.05749
    echo:  G29 W I3 J3 Z0.99999
    echo:  G29 W I4 J3 Z0.95749
    echo:  G29 W I0 J4 Z1.19249
    echo:  G29 W I1 J4 Z1.17999
    echo:  G29 W I2 J4 Z1.11749
    echo:  G29 W I3 J4 Z1.08499
    echo:  G29 W I4 J4 Z1.02749
    echo:; Material heatup parameters:
    echo:  M145 S0 H200.00 B60.00 F255
    echo:  M145 S1 H240.00 B70.00 F255
    echo:; PID settings:
    echo:  M301 P21.73 I1.54 D76.55
    echo:; Retract: S<length> F<units/m> Z<lift>
    echo:  M207 S3.00 W13.00 F4800.00 Z0.30
    echo:; Recover: S<length> F<units/m>
    echo:  M208 S0.00 W0.00 F4800.00
    echo:; Z-Probe Offset (mm):
    echo:  M851 X-45.00 Y-7.00 Z-3.30
    echo:; Filament load/unload lengths:
    echo:  M603 L415.00 U450.00
    echo:; Filament runout sensor:
    echo:  M412 S0 D8.00
    ok P15 B3

**Reference: The start of my gcode file**

    ;FLAVOR:Marlin
    ;TIME:775
    ;Filament used: 0.480677m
    ;Layer height: 0.2
    ;MINX:16.516
    ;MINY:16.515
    ;MINZ:0.2
    ;MAXX:218.485
    ;MAXY:218.485
    ;MAXZ:0.2
    ;Generated with Cura_SteamEngine 4.9.0
    M140 S60
    M105
    M190 S60
    M104 S215
    M105
    M109 S215
    M82 ;absolute extrusion mode
    ; Ender 3 Custom Start G-code
    G92 E0 ; Reset Extruder
    G28 ; Home all axes
    G29 ; Auto bed levelling
    G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
    G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
    G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
    G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
    G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
    G92 E0 ; Reset Extruder
    G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
    G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
    G92 E0
    G92 E0
    G1 F2700 E-5
    ;LAYER_COUNT:1
    ;LAYER:0
    M107
    G0 F6000 X26.14 Y20.098 Z0.2
    ;TYPE:SKIRT
    G1 F2700 E0

**Reference: Congifuration.h**

As I've used two precompiled firmwares (see point 4 above), I don't have a `Congifuration.h` to share. But I could compile my own version of the firmware if needed.

**Concluding**

I hope I've given a detailed enough account for you guys to help me. If you have any additional questions I'll try to answer them as quickly as possible. Thanks a million!

ellensp, 2021-05-13:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Zjaaspoer, 2021-05-13:
@ellensp ok thanks for the very quick response! 🙏 I'll get to following these exact steps asap. I will probably be able to do these this weekend and will report back as soon as I have all the information requested! 💪

Zjaaspoer, 2021-05-19:
@ellensp apologies, it took a bit longer than I hoped. But I've followed the exact instructions.

Please find the terminal output attached.

I've also attached the changes I made to `Configuration.h` and `Configuration_adv.h` compared to the default configuration provided for the Ender 3

[terminal output](https://github.com/MarlinFirmware/Marlin/files/6510607/terminal.txt)
[configuration diff](https://github.com/MarlinFirmware/Marlin/files/6510606/configuration_diff.txt)

I've got the pipeline to build the firmware all set up, so if you have any additional requests please let me know and I'll be able to respond pretty quickly this time.

Zjaaspoer, 2021-07-19:
Just a quick rely to keep the issue open, hoping on a way to resolve this issue. If there is any information missing or anything I can try please let me know!

XDA-Bam, 2021-07-20:
This might be fixed by 154decf. Please check out the latest bugfix-2.0.x.

Zjaaspoer, 2021-07-23:
Just tried this out with the freshest version of bugfix-2.0.x today. Unfortunately, the problem persisted.

nils-nils, 2021-08-30:
I had similar problems until a week ago - I actually had multiple problems and gave up on the printer for a couple months before first figuring out how to get probing to work reliably (no more problems deploying and retracting the pin and nozzle crashes and random error code generation) only to have the Z-offset needing constant adjustment - at which point I corrected some mechanical problems with the printer:

1. adjusted the vertical beams of the frame (same distance apart at the bottom and top)
2. checked that the gantry was the same distance from the base of the printer at the left and right sides
3. checked and tightened the eccentric nuts for each axis

These videos helped me figure this out https://www.youtube.com/watch?v=JTN6jtB5mqk&t=1365s and https://www.youtube.com/watch?v=P5HJgIrgGXc.

BL-Touch is now providing reproducible results and leveling is working as I had always hoped.

I also modified the startup gcode for the printer a little in cura, tried several different firmware versions, used modified configuration files for marlin, messed around with bits of marlin, tried UBL, printed all sorts of calibration tests, cussed and cursed and wrote complaints to creality. Not sure how much of that was necessary but I had no idea where the problems were coming from...

lloydbigoil, 2021-09-24:
This problem still persists. I even watched the videos took apart my printer and re built using the same tools they did in the video. taking careful note to make things square and level. Checked all my belts and bolts. printer shows almost level in octoprint but 1st layers never stick, ive used glass and PEI sheets. I held my stepper motor to feel it move around. start code is : 

```
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M420 S1;
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M420 S1;
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
```

hamster65, 2021-12-02:
Still a problem for me. I get far better results with ABL/UBL disabled. I think I can rule out machanical issues because levelling works fine for me with Klipper on the same hardware. I think Marlin generates too much movement on the z axis.

Zjaaspoer, 2022-02-01:
This problem unfortunately still exists.

tombrazier, 2022-03-04:
@Zjaaspoer I think multiple things are going on here:

1. Your bed appears to be quite sloped. In addition to what @nils-nils suggests, the tramming wizard (ASSISTED_TRAMMING) should help you to adjust the heights of you bed corners to reduce the slope (that's assuming it works with the Ender LCD). However, regardless of this bed leveling still appears not to be working correctly for you. 

2. Your terminal log lines 901 and 902 has:
```
Recv: G29 uncorrected Z:13.30
Recv:  corrected Z:13.30
```
Those values should not be the same. The second one should be the corrected Z value. Do you have an all zero mesh stored in EEPROM? #23843, which I am working on now, may be responsible for this and would result in a general incorrect Z position across the whole bed. Otherwise for some reason bed leveling is not being applied at all.

It would be good to see the output of `M420 V` and `M114 D` at X200 Y200 and X0 Y0 after `G29`.

3. Your bed Z offsets are quite a long way from zero. With a fade height of 10mm and layer height of 0.2mm, there is already 1/50th of the levelling that does not get applied. To a small degree this will nullify the effects of the leveling and to a larger degree the average bed height will have a more significant effect.

Even odder, they are positive values - around 1mm in your original post and between 0.108mm and 0.535mm in the log you posted. I take it you have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN. I would have expected the Z values for bed levelling to be quite close to zero because they are measured with the same probe as Z homing.

It might be worth looking into this. Also perhaps play with MANUAL_Z_HOME_POS.

4. I don't know whether the Ender 3 is vulnerable to X axis twist. But it might be worth trying the new X axis twist functionality (X_AXIS_TWIST_COMPENSATION). Again I don't know whether the Ender UI makes the wizard available.

descipher, 2022-03-04:
Could be a corrupt EEPROM.DAT file or nonfunctional EEPROM in flash.

Please enable this in configuration.h and retry.
I have the 4.2.7 running with Bilinear ABL and a BLTouch with no issues.

#define RESTORE_LEVELING_AFTER_G28
#define SDCARD_EEPROM_EMULATION

#define EEPROM_SETTINGS       // Persistent storage with M500 and M501 //BG
#if ENABLED(EEPROM_SETTINGS)
  #define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.
  #define EEPROM_INIT_NOW   // Init EEPROM on first boot after a new build.
#endif

Zjaaspoer, 2022-03-05:
Wow, thanks for the great detailed feedback. Will implement all your suggestions and report back asap.

Thanks again.

Zjaaspoer, 2022-05-05:
Still have to look into [this](https://github.com/MarlinFirmware/Marlin/issues/21887#issuecomment-1059081054) comment.

closed 2022-07-15 by github-actions[bot]