MarlinFirmware/Marlin #21893: In response to bug 21875. Compiles but doesn't flash to board. Board stuck at Flash complete.

rclarke250, 2021-05-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With the new pins file for MKS Robin Pro, I get 2 errors compiling fir lvgl and Robin tft35 display. 
Marlin\src\lcd\extui\mks_ui\printer_operation.cpp: In function 'void filament_check()':
Marlin\src\lcd\extui\mks_ui\printer_operation.cpp:161:13: warning: unused variable 'FIL_DELAY' [-Wunused-variable]
   const int FIL_DELAY = 20;

Compiling .pio\build\mks_robin_pro\libd5a\lvgl\lv_draw\lv_draw_arc.c.o
.pio\libdeps\mks_robin_pro\lvgl\src\lv_core\lv_obj.c: In function 'lv_deinit':
Compiling .pio\build\mks_robin_pro\libd5a\lvgl\lv_draw\lv_draw_basic.c.o
.pio\libdeps\mks_robin_pro\lvgl\src\lv_core\lv_obj.c:130:5: warning: implicit declaration of function 'lv_log_register_print_cb'; did you mean 'lv_obj_get_event_cb'? [-Wimplicit-function-declaration]
     lv_log_register_print_cb(NULL);
     ^~~~~~~~~~~~~~~~~~~~~~~~
     lv_obj_get_event_cb

This still compiles with  a success message. flash to the board, it gets stuck at 100% flash complete, never updates the lvgl assets, and when I look at the SD card afterward, it now has an eprom.dat file that was never there.

### Bug Timeline

new, related to 21875

### Expected behavior

Flash to complete and the board to reboot with icons.

### Actual behavior

board failed to flash properly

### Steps to Reproduce

fresh download of Marlin bugfix. setup environment to mks_robin_pro
adjust config.h for board_mks_robin_pro
import config.h from examples from bugfix configurations, or build fresh
choose mks_robin_tft35 as display
choose lvgl_ui
choose touch screen
compile
attempt to flash

### Version of Marlin Firmware

2.08 bugfix latest at 6:30am 5/13

### Printer model

JGMaker Artist-D Pro

### Electronics

stock. mks_robin_pro with mks_robin_tft35, tmc2209_standalone

### Add-ons

none

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

This is an IDEX machine. Disabled filament runout, even changing the script to use M25 instead of M600 causes an issue.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6472357/Marlin.zip)

rhapsodyv, 2021-05-13:
Those warnings don’t hang the board.

it only happen with lvgl ui? 

There’s high possibility that lvgl is incompatible with this board. I don’t have it and I can’t test. Lvgl ui needs a special spi flash chip, that I don’t know if this board have it.

Worst yet: probably lvgl ui is incompatible with idex machine at this time.

Test with color ui. If it works, keep using it.

And this will become a feature request to make lvgl work with your setup.

rclarke250, 2021-05-13:
doesn't work in classic, that flashes, but the display is mirrored. And it doesn't work in Color UI either, compiles with no errors, but will not boot. blank screen, and power supply has an alarm and is looping.The Original firmware (Marlin 2.05) has a firmware.bin file, along with 2 folders, mks_font which has a FontUNIGBK.bin file, and another folder called mks_pics and has the bmp_XXX.bin files. and since it was a FSMC display, I chose lvgl_ui.  Not sure if there is a way to get a new 2.08 firmware to use the custom xxx.bin files supplied with the factory firmware.  The only hardware change I made from factory, is select uart for tmc2209 instead of standalone, so just a jumper change on the board.

rhapsodyv, 2021-05-13:
For mirrored display, you can use the TFT_ROTATION config to fix it.

for the hang, only that display stays black? The serial communication works?

rclarke250, 2021-05-13:
For color_ui it stays black, with an alarm on my power supply. Fan and red
led pulsing instead of staying on. I do not know why there is an eprom.dat
file on my ad card after attempting the flash. Never had that before.

rhapsodyv, 2021-05-13:
It’s seems there’re more issues than the TFT.  It worth you trying to disable configs until you find the main problem.

rclarke250, 2021-05-13:
Alright, I downloaded the latest version, and configuration examples from
the bugfix branch. Setup PlatformIO for mks_robin_pro. Imported the
configuration file from the examples. Set the driver type to
TMC2209_standalone  did not add the other drivers, or make changes to
direction. Set Robin_tft35 for display,lvgl_ui and enabled touch landscape.
Compiled...get 3 warnings...flash my 3d printer....says TFT updating 100%
and then stops, been that way for over 5 minutes. Still getting that
epro.dat file on the sdcard. Cold boot says Booting..... never goes any
further.

Next test, changed lvgl_ui to Color_ui, no other changes. compiles no
errors. TFT updating 100% and then stops... Cold Boot says Booting.... and
hangs there. Also get an eprom.dat file along with firmware.cur on sd card.

Next test, changed to Classic ui no other changes. compiles with no errors.
updates, but display is mirrored, and no touch functions, eprom.dat on sd
card again.
Flash back to factory, and it works like before. Color was at least working
before, so not sure what is going on.

rhapsodyv, 2021-05-13:
Try this branch: https://github.com/rhapsodyv/Marlin/tree/robin-pro-tft-fix

rclarke250, 2021-05-13:
okay, trying it now.

rclarke250, 2021-05-13:
that is better, lvgl now compiles, and I can flash, screen is backwards and
mirror, so I am going to try to rotate.

rclarke250, 2021-05-13:
it is ignoring the TFT rotate setting....touch is correct, butit is
mirrored on the X axis. I define TFT_MIRROR_X and it looks the same as if I
don't set the rotate. still getting the eprom.dat file on the sdcard also.

rhapsodyv, 2021-05-13:
Attach a picture

rclarke250, 2021-05-13:
I tried color ui. This is with define rotate mirror x enabled. Looks normal
when updating firmware, and when first turned on booting.... Looks normal.
As soon as the logo comes up it flips.

rhapsodyv, 2021-05-13:
Send me some pictures of the display. Are you sure it is 3.5"? 

It's not 3.2" MKS_ROBIN_TFT_V1_1R ?

rclarke250, 2021-05-13:
Here is the back showing the model.

rhapsodyv, 2021-05-13:
Here where?

rclarke250, 2021-05-13:
[image: IMG_20210422_082019074.jpg]
weird that it didn't send from my phone.

rhapsodyv, 2021-05-14:
You can try send it using discord. But the image didn't show here yet...

thisiskeithb, 2021-05-14:
> weird that it didn't send from my phone.

Replying by email strips out a lot of stuff. Replying on the web/via browser directly (https://github.com/MarlinFirmware/Marlin/issues/21893) will work better.

thinkyhead, 2021-05-14:
Patch merged. Please close this PR if it has fixed all issues.

PR #21900 (2021-05-14): Robin Pro TFT Fixes (2)
Finish the fixes of TFT with Robin pro.

The first PR (#21879) was merged before user test it.

Now he tested and everything works.

Fix: #21893
closed 2021-05-16 by rhapsodyv