MarlinFirmware/Marlin #21895: [FR] Monoprice MP10 / Malyan MA 10

jipp, 2021-05-13:
### Is your feature request related to a problem? Please describe.

HI

I'm looking for the correct settings for the above printers. So far I found only small hints, like the printer use STM32 Cortex M4 processors.

Are the printers above supported by Marlin 2.x?

any help/tips appreciated.

thanks

### Are you looking for hardware support?

Monoprice MP10 / Malyan MA 10

### Describe the feature you want

looking for the config/board settings for the above printer

thisiskeithb, 2021-05-14:
Do you have pictures of the motherboard & LCD/TFT screen?

ellensp, 2021-05-14:
A bit of googling shows
display and ESP control board - https://www.monoprice.com/product?p_id=39107
Stepper heater board - https://www.monoprice.com/product?p_id=39106

The main controller chip is under that capacitor, rather unreadable.

jipp, 2021-05-14:
From What I found by googleing:

https://www.reddit.com/r/3Dprinting/comments/dqu8fb/please_help_me_identify_my_3d_printer_mainboard/

so the board should be a ARM Cortex M4 STM32F401RBT6

would this help?

thisiskeithb, 2021-05-14:
> so the board should be a ARM Cortex M4 STM32F401RBT6
> 
> would this help?

It does.

I can't find any existing board or MCU support in Marlin currently.

ellensp, 2021-05-14:
Basically nothing is known about this board or the LCD, or the protocol between the two.
This would need serious effort to firstly document all the pins on the controller, then reverse engineer the protocol the LCD side uses. 
Has this been done?. No
Can it be done?  Given time with someone who has this machine and the interest and capabilities, probably, but not any time soon.

jipp, 2021-05-15:
Hi

thx a lot for the fast answer. Saved me a lot of time for investigations.

all the best and stay healthy

closed 2021-05-15 by jipp