MarlinFirmware/Marlin #21896: [BUG] BLTOUCH doesn't deploy and Error:Probing Failed/endstops hit:  Z:"value"

ThePsyk8, 2021-05-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

At the first start I set z-offset, after I do autohome from Pronterface or from printer and it works. M48 also works.
The problem is when I want to do the automatic bed levelling, the bltouch doesn't deploy at first probe and I receive an error: Error:Probing Failed. Otherwise BLTOUCH makes some probes and at a certain point it stop working and I receive an error:
STOPPED or endstops hit:  Z:"value".
If I start a print from SD or slicer, it starts but when the BLTOUCH gives the error, the print continues without compensating.

https://user-images.githubusercontent.com/65788763/118170632-9cf72200-b42a-11eb-8662-52a99a59122f.mp4

### Bug Timeline

From Marlin 2.0.7.2

### Steps to Reproduce

- Power on printer;
- Home all axis;
- Set z-offset;
- Once all axis are homed, send a G29 command from Pronterface/printer;

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

Alfawise U30 (motherboard: V08)

### Electronics

STM32F103VE_longer (V08)

### Add-ons

BLTOUCH

### Your Slicer

IdeaMaker

### Host Software

Pronterface

### Additional information & file uploads

echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
>>> g28
SENDING:G28
echo:G28
>>> G28  X0.00 Y0.00 Z0.00
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X-35.00 Y-5.50 Z-0.50 (Left-Front & Below Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
Raise Z (before homing) by 5.00
>>> do_blocking_move_to  X0.00 Y0.00 Z0.00
>  X0.00 Y0.00 Z5.00
<<< do_blocking_move_to  X0.00 Y0.00 Z5.00
BLTouch Command :160
BLTouch Command :90
last_written_mode - 0
config mode - OD
[... 312 lines cut ...]
FAST Probe fail! - No trigger.
<<< Probe::run_z_probe  X177.50 Y20.50 Z-1.50
  current_position= X177.50 Y20.50 Z-1.50 : Probe::set_deployed
deploy: 0
Probe::do_z_raise(10.00)
>>> do_blocking_move_to  X177.50 Y20.50 Z-1.50
>  X177.50 Y20.50 Z10.50
echo:busy: processing
<<< do_blocking_move_to  X177.50 Y20.50 Z10.50
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
>>> do_blocking_move_to  X177.50 Y20.50 Z10.50
>  X177.50 Y20.50 Z10.50
<<< do_blocking_move_to  X177.50 Y20.50 Z10.50
Error:Probing Failed
[ERROR] Error:Probing Failed

<<< Probe::probe_at_point  X177.50 Y20.50 Z10.50
  current_position= X177.50 Y20.50 Z10.50 : Probe::set_deployed
deploy: 0
  current_position= X177.50 Y20.50 Z10.50 : > probing complete
<<< G29  X177.50 Y20.50 Z10.50

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6474403/Configuration.zip)

ellensp, 2021-05-14:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ThePsyk8, 2021-05-14:
[M111 S247.txt](https://github.com/MarlinFirmware/Marlin/files/6476684/M111.S247.txt)

Thanks

ThePsyk8, 2021-05-21:
Hello,
Is there a solution?
Thanks

closed 2021-08-01 by github-actions[bot]