MarlinFirmware/Marlin #21904: [BUG] Neither PREHEAT_*_LABEL nor PREHEAT_3_* recognized

tefracky, 2021-05-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am using an Ender 3 V2 and tried to customize the preheat settings. The preheat label using PREHEAT_*_LABEL is not changed in the display. The temperature and fan speed is changed and used, but the label not changed. Additionally, I tried to add a third filament, but it isn't shown up at all.

### Bug Timeline

I only tried 2.0.7.2, 2.0.8 and 2.0.x-bugfix

### Expected behavior

Label and third filament should be shown correctly.

### Actual behavior

Label not changed and third filament not shown.

### Steps to Reproduce

1. Change PREHEAT_*_LABEL
2. Add PREHEAT_3_*
3. Compile and install
4. Take a look on the display

### Version of Marlin Firmware

2.0.x-bugfix at commit 2a5ab55c36380cd394c4922705412452f800c17c from 14.05.2021

### Printer model

Ender 3 V2

### Electronics

Stock

### Add-ons

Not related

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

configuration.h
[Marlin-bugfix-2.0.x-UBL.zip](https://github.com/MarlinFirmware/Marlin/files/6477508/Marlin-bugfix-2.0.x-UBL.zip)

```
#define PREHEAT_1_LABEL       "PLA"
#define PREHEAT_1_TEMP_HOTEND 195
#define PREHEAT_1_TEMP_BED     66
#define PREHEAT_1_TEMP_CHAMBER 35
#define PREHEAT_1_FAN_SPEED   0 // Value from 0 to 255

#define PREHEAT_2_LABEL       "PETG"
#define PREHEAT_2_TEMP_HOTEND 220
#define PREHEAT_2_TEMP_BED     81
#define PREHEAT_2_TEMP_CHAMBER 35
#define PREHEAT_2_FAN_SPEED   0 // Value from 0 to 255

#define PREHEAT_3_LABEL        "Wood"
#define PREHEAT_3_TEMP_HOTEND  195
#define PREHEAT_3_TEMP_BED     66
#define PREHEAT_2_TEMP_CHAMBER 35
#define PREHEAT_3_FAN_SPEED    0 // Value from 0 to 255
```

ellensp, 2021-05-14:
Ender 3 V2 uses a custom display with its own firmware.
Marlin setting are not used within it

tefracky, 2021-05-14:
> Ender 3 V2 uses a custom display with its own firmware.
> These Marlin setting are not used within this type of display.

Thanks! Do you know where I can change it?

ellensp, 2021-05-14:
you would need to change the firmware inside the lcd. 
Ie create your own DWIN_SET
In https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2/DWIN_SET 
Is a file 9.ICO, this file actually contains all icons used in the display. 
There are only two preheat icons, PLA and ABS
[image]
[image]
If you search for ICON_PLAPreheat and ICON_ABSPreheat you will find the marlin code that displays them.

closed 2021-07-24 by github-actions[bot]