MarlinFirmware/Marlin #21908: [BUG] Compilation error when MESH_EDIT_MENU is enabled for AUTO_BED_LEVELING_[LINEAR|3POINT]

ondrada, 2021-05-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When ABL is configured as AUTO_BED_LEVELING_LINEAR or AUTO_BED_LEVELING_3POINT and both LCD_BED_LEVELING and MESH_EDIT_MENU options are enabled in Configuration.h, Marlin fails to compile and gives errors about undeclared preprocessor variables Z_VALUES (for both AUTO_BED_LEVELING_3POINT and AUTO_BED_LEVELING_LINEAR) and GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y (for AUTO_BED_LEVELING_3POINT only).

I've tried to solve that by adding a condition to enable Mesh edit menu only if HAS_MESH (here: https://github.com/ondrada/Marlin/commit/63150c5e0c2f404f0b2856ea2a863a40c6af2e59 ). But don't know if it wouldn't be better to add a conditional like "MESH_EDIT_MENU is not compatible with AUTO_BED_LEVELING_LINEAR/AUTO_BED_LEVELING_3POINT", or if it is possible to have the menu without 'Z' part for AUTO_BED_LEVELING_LINEAR.

### Expected behavior

Compilable Marlin when AUTO_BED_LEVELING_[LINEAR|3POINT] is selected together with LCD_BED_LEVELING and MESH_EDIT_MENU, or some incompatibility message.

### Actual behavior

Compilation failure with beyond combination of options in Configuration.h

### Steps to Reproduce

Try to compile Marlin with:
LCD_BED_LEVELING enabled
MESH_EDIT_MENU enabled
AUTO_BED_LEVELING_LINEAR or AUTO_BED_LEVELING_3POINT enabled

### Version of Marlin Firmware

2.0.x bugfix

### Printer model

Ender 3 Pro

### Electronics

BTT E3 RRF

ellensp, 2021-05-15:
AUTO_BED_LEVELING_LINEAR and AUTO_BED_LEVELING_3POINT are not MESHES, so enabling MESH_EDIT_MENU does not make sense.

closed 2021-05-15 by ellensp

ondrada, 2021-05-15:
OK, then shouldn't there be some message like "MESH_EDIT_MENU is not compatible with AUTO_BED_LEVELING_LINEAR/AUTO_BED_LEVELING_3POINT" in SanityCheck.h or somewhere, rather than compile errors at Marlin/src/lcd/menu/menu_bed_leveling.cpp ?