MarlinFirmware/Marlin #21914: [BUG] Mini Rambo board flashing firmware causes halt

brotherchris, 2021-05-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a machine running a mini Rambo board and a 2004 screen. I have been running Marlin 1.1.9 for a long time and decided to upgrade to 2.0.8. After the upgrade, if I have EXP2 plugged in for the screen, when marlin boots, the printer goes right to halt. The terminal reports kill pin hit.

Error:!! KILL caused by KILL button/pin
Error:Printer halted. kill() called!

With EXP2 unplugged it will run, but of course I lose the ability to scroll through the menus. 

I can flash back to 1.1.9 and everything works fine. I have tried 2.0.7.2, 2.0.8 and bugfix as of 5/14/21 and all have the same behavior. I am using the Rambo ENV in VScode to build the firmware. If I use the mega2560 ENV to build, the same thing happens. 
[Riley 2.0.8 Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6486429/Riley.2.0.8.Marlin.zip)

### Bug Timeline

This appears to be an older issue. Not sure when it appeared in Marlin.

### Expected behavior

To be able to use the EXP2 connection on the LCD as I did in 1.1.9.

### Actual behavior

When EXP2 is plugged in kill pin is hit and printer halts. 

### Steps to Reproduce

When using the Mini Rambo board and 2004 screen, flash to Marlin 2.0.8, the issue will appear. 

### Version of Marlin Firmware

2.0.8

### Printer model

Prusa clone 

### Electronics

Ultimachine Mini Rambo

### Add-ons

N/A

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

ellensp, 2021-05-15:
Perhaps your reset button is broken, stick on, and the old firmware ignored it? It is a IO pin not a real reset.

brotherchris, 2021-05-15:
@ellensp I can do some poking around with the multi meter and check a few things, not a bad idea. This actually happens with multiple screens and the reset button does work to reset it. Unfortunately, when it resets it comes right back to halt.

ellensp, 2021-05-15:
If reset works its not that..
In Marlin/src/pins/rambo/pins_MINIRAMBO.h
Comment out "#define KILL_PIN                          32"
See if it likes that

brotherchris, 2021-05-15:
@ellensp Yes, commenting ou tthe kill pin fixes it.

thinkyhead, 2021-05-16:
852a8d6685ec1137eb65e78fa748cae41fbd36b6 Configurable KILL_PIN_STATE

thinkyhead, 2021-05-16:
Does the PR linked above make sense to y'all?

closed 2021-06-04 by thinkyhead