MarlinFirmware/Marlin #21916: [BUG] Compile error with G29_RETRY_AND_RECOVER defined and BLTouch disabled

TheFiZi, 2021-05-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling 2.0.8.1 and bugfix-2.x with the BLTouch disabled and `G29_RETRY_AND_RECOVER` enabled the build fails

This did not occur previously with any 2.x firmware I've built.

I think it's this bug back from the dead?: https://github.com/MarlinFirmware/Marlin/issues/20850

Config files
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/6488329/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/6488330/Configuration_adv.txt)

I am using these as a base configuration: https://github.com/MarlinFirmware/Configurations/tree/release-2.0.8.1/config/examples/Creality/CR-10%20V3 and making some very minor tweaks (BLTouch enable/disable, Baud rate, some version strings and a logo)

If I change `#define G29_RETRY_AND_RECOVER` to `//#define G29_RETRY_AND_RECOVER` the firmware will build.

Output from build job:

```
PS D:\Downloads\Marlin-2.0.x> platformio run -e mega2560
Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (2.2.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.0.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.4
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
Compiling .pio\build\mega2560\src\src\core\utility.cpp.o
Compiling .pio\build\mega2560\src\src\feature\cancel_object.cpp.o
Compiling .pio\build\mega2560\src\src\feature\e_parser.cpp.o
Compiling .pio\build\mega2560\src\src\feature\host_actions.cpp.o
Compiling .pio\build\mega2560\src\src\feature\pause.cpp.o
Compiling .pio\build\mega2560\src\src\feature\powerloss.cpp.o
Compiling .pio\build\mega2560\src\src\feature\runout.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M304.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M309.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M111.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M120_M121.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M17_M18_M84.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M211.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M400.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M80_M81.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M85.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M993_M994.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M997.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M999.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\T.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\eeprom\M500-M504.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\cancel\M486.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\digipot\M907-M910.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\pause\G27.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\pause\M125.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\pause\M600.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\pause\M603.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\pause\M701_M702.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\powerloss\M1000.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\powerloss\M413.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\feature\runout\M412.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\gcode.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\gcode_d.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\geometry\G92.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\host\M110.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\host\M113.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\host\M114.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\host\M115.cpp.o
Marlin\src\gcode\gcode.cpp:216:40: error: no 'void GcodeSuite::event_probe_recover()' member function declared in class 'GcodeSuite'
   void GcodeSuite::event_probe_recover() {
                                        ^
Compiling .pio\build\mega2560\src\src\gcode\host\M118.cpp.o
Marlin\src\gcode\gcode.cpp:230:40: error: no 'void GcodeSuite::event_probe_failure()' member function declared in class 'GcodeSuite'
   void GcodeSuite::event_probe_failure() {
                                        ^
Marlin\src\gcode\gcode.cpp:249:35: error: no 'void GcodeSuite::G29_with_retry()' member function declared in class 'GcodeSuite'
   void GcodeSuite::G29_with_retry() {
                                   ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mega2560\src\src\gcode\gcode.cpp.o] Error 1
============================================================================================== [FAILED] Took 16.74 seconds ============================================================================================== 

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:16.744
========================================================================================= 1 failed, 0 succeeded in 00:00:16.744 ========================================================================================= 
```

### Bug Timeline

New as of 2.0.8.1 for me at least. I've previously compiled 2.0.6 -> 2.0.8.0 with out issue

### Expected behavior

Firmware should build

### Actual behavior

Firmware fails to build

### Steps to Reproduce

1. Copy/paste configuration files into Marlin source directory
2. Compile firmware

### Version of Marlin Firmware

2.0.8.1

### Printer model

CR-10 V3

### Electronics

Stock

### Add-ons

BLTouch v3.1

### Your Slicer

IdeaMaker

### Host Software

OctoPrint

ellensp, 2021-05-16:
This PR should fix it

TheFiZi, 2021-05-16:
Yup that did it.

[build.log](https://github.com/MarlinFirmware/Marlin/files/6490160/build.log)

TheFiZi, 2021-06-05:
This bug has returned in 2.8.2.0.

cp2004, 2021-06-05:
The mentioned PR has not yet been merged...

TheFiZi, 2021-06-05:
> 
> 
> The mentioned PR has not yet been merged...

Gotcha

thisiskeithb, 2021-06-06:
PR's been merged. Pull a fresh copy of `bugfix-2.0.x` or wait for the next tagged release.

closed 2021-06-06 by thisiskeithb