MarlinFirmware/Marlin #21917: [BUG] Z_HOMING_HEIGHT not raising Z

neilvangeffen, 2021-05-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I cannot get Z to raise before homing. I have the settings right as far as i can tell, and the dubus says its going to raise Z, but it doesnt actually raise Z

### Bug Timeline

Doesnt work in 2.0.7.2, 2.0.8, or bugfix (havent tried others)

### Expected behavior

I expect Z to raise when I home any axis if Z is below Z_HOMING_HEIGHT, or if Z is unknown after reboot, or motors disabled

### Actual behavior

it says it will raise Z but doesnt, and the first move it does, is home X

### Steps to Reproduce

1. Turn on Machine
2. G28

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 12573ddDL bugfix-2.0.x (May 15 2021 21:37:43) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Anet A6 12573ddDL EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Anet A6

### Electronics

SKR 1.4 Turbo, TMC2209

### Add-ons

independent Z drivers, Sensorless homing

### Your Slicer

Simplify3D

### Host Software

Same as my slicer

### Additional information & file uploads

- The debug says its doing the right thing, but it doesnt physically move. I even set it to go to 15mm so there was NO WAY i'd miss it... but nothing.
`G28 X0 Y0 Z0
READ: >>> G28  X-2.00 Y0.00 Z0.00
READ: Machine Type: Cartesian
READ: Probe: FIX_MOUNTED_PROBE
READ: Probe Offset X0.00 Y0.00 Z0.00 (Aligned With & Same Z as Nozzle)
READ: Auto Bed Leveling: BILINEAR (disabled)
READ: 
READ: Raise Z (before homing) by 15.00
READ: >>> do_blocking_move_to  X-2.00 Y0.00 Z0.00
READ: >  X-2.00 Y0.00 Z15.00
READ: <<< do_blocking_move_to  X-2.00 Y0.00 Z15.00
READ: >>> homeaxis(X)`

- If i turn on the machine and send a G0 Z15 move, it works fine and clearly moves.
- the printer works. I've printed a number of models with it, it just doesnt do the Z raise as of yet
- note, it has bed level and probe enabled in the code, but i havent used any of that as of yet

and a video for those that dont believe me https://photos.app.goo.gl/ERgyD3Bw1JM2mq719

[Z No Raise.zip](https://github.com/MarlinFirmware/Marlin/files/6488319/Z.No.Raise.zip)

neilvangeffen, 2021-05-19:
I really wish someone could point me at a place to look...

ellensp, 2021-05-19:
This warning going past your screen 151 times didn't make you wonder at all??
```
In file included from Marlin/src/sd/../inc/../pins/pins.h:796,
                 from Marlin/src/sd/../inc/MarlinConfig.h:34,
                 from Marlin/src/sd/cardreader.cpp:23:
Marlin/src/sd/../inc/../pins/pins_postprocess.h:744: warning: "Z2_USE_ENDSTOP" redefined
  744 |       #define Z2_USE_ENDSTOP _YMAX_
      | 
In file included from Marlin/src/sd/../inc/MarlinConfigPre.h:56,
                 from Marlin/src/sd/../inc/MarlinConfig.h:28,
                 from Marlin/src/sd/cardreader.cpp:23:
Marlin/src/sd/../inc/../../Configuration_adv.h:661: note: this is the location of the previous definition
  661 |     #define Z2_USE_ENDSTOP          _XMAX_
```
comment out  #define Z2_USE_ENDSTOP          _XMAX_ in Configuration_adv.h 
This is not needed when an E is reused as a Z, its is done for you.

Secondly  sensorless homing on lead screws is a really bad idea.  This will end in holes in your bed.
There is insufficient downwards movement per rotation to trigger the sensor less sensing reliability. 

The safe way to home Z is with a probe or real end stops.

ellensp, 2021-05-19:
you have 
#define USE_ZMAX_PLUG this would stop all up movement, comment it out. and you dont have anything plugged in there.

closed 2021-05-19 by ellensp