MarlinFirmware/Marlin #21918: [BUG] (bug summary)

Sutan24, 2021-05-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi there, Well Marlin 2.0.8 having a problem regards to the power loss recovery matter.
During Power loss:

1. POWER_LOSS_ZRAISE 5: Z-axis not raising (nothing happening when power loss)
2. POWER_LOSS_RETRACT_LEN 10: is not retracting (nothing happening when power loss)

When power resumes: 
X & Y axis moves filament purge working fine as assigned, continue printing 

My controller board : BIGTREE_SKR_PRO 1.1
Printing method: SD
Slicer: Cura

I have a custom-built UPS 12V that can be programmed to make it LOW or HIGH power loss signals when a power failure occurs. Has a precision intelligence that signals power loss. Was working absolutely fine with 2.0.

I spend a lot of time changing both HIGH and LOW power loss states. And tried Pullup, Pulldown, and un command both. Assigned different power loss pins.

Power loss recovery works like there is no pin assigned. 

I measured the voltages when changing pullup/down/un commended.  It’s working accordingly.

My guess is the power loss pin not working with the firmware.

Also tried to enabled  #define POWER_LOSS_RECOVER_ZHOME when compile msg throwing an error: cannot be used with Z safe Homing

Moreover tried  #define PLR_ENABLED_DEFAULT  tried both false & true

Also, tried sending commands manually  M413 S1 & M413 S0 with no luck at all.

Any help would be appreciated. 

### Bug Timeline

new

### Expected behavior

When Power loss occurs

1. POWER_LOSS_ZRAISE: Z-axis should raise as assigned (when power loss pin senses power loss)
2. POWER_LOSS_RETRACT_LEN: should retract as assigned (when power loss pin senses power loss)

### Version of Marlin Firmware

2.0.8

### Host Software

Cura

### Additional information & file uploads

[

Sutan24, 2021-05-16:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6488446/Marlin.zip)

ellensp, 2021-05-16:
Please update issue subject

closed 2021-07-28 by github-actions[bot]