MarlinFirmware/Marlin #21923: [BUG] Encoder button CR10 display + SKR V1.4T not responding

Emmaruk79, 2021-05-16:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Encoder push button doesn't respond 
Encoder (rotation) works, you can choose/select items but not confirm it.
Accoured in 2.0.8 and in 2.0.8.1 flashed back to 2.0.7.2 everything works.

### Bug Timeline

Afther install of 2.0.8 and still in 2.0.8.1

### Expected behavior

comfirm selected menu item

### Actual behavior

rotation of encoder button works
push / confirm doesn't work

### Steps to Reproduce

flashed 2.0.8: button not working
flashed 2.0.7.2 button working
flashed 2.0.8.1 button not working

No alterations made to the hardware

### Version of Marlin Firmware

2.0.8.1

### Printer model

Creality Ender 5 Plus

### Electronics

Board used: BTT SKR V1.4 Turbo (connector used EXP1) Display: Creality CR10 stockdisplay (connector used EXP3)

ellensp, 2021-05-16:
Please attach your configuration files.

Emmaruk79, 2021-05-16:
[Marlin config files.zip](https://github.com/MarlinFirmware/Marlin/files/6488880/Marlin.config.files.zip)

the config files in a .ZIP

ellensp, 2021-05-16:
I just trued this on current bugfix, same hardware. only  setting CR10_STOCKDISPLAY  and MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO and default_envs = LPC1769 and it works fine
LCD knob rotation and selection

ellensp, 2021-05-16:
Definitely something in your config as can replicate with your config files.

Emmaruk79, 2021-05-16:
okay, thanks for the comfimation, got some work/checking to do

ellensp, 2021-05-16:
You have enabled #define NO_LCD_MENUS
So the button doesn't do anything as there are no menus to go to!

closed 2021-05-16 by ellensp

Emmaruk79, 2021-05-16:
just found it my self. 
i feel stupid now.
sorry