MarlinFirmware/Marlin #21926: [FR] variable speed, acceleration, and jerk depending on axis for cartesian printers

yet-another-average-joe, 2021-05-16:
### Is your feature request related to a problem? Please describe.

With cartesian printers, we can see variable hosting/ringing depending on the direction. This is obvious when looking at a text embossed along a vertical and cylindrical/conical surface.

### Are you looking for hardware support?

Could be usefull to all cartesian printers, except those with Ultimaker kinematics ; All printers that are accelerating different masses depending on the axis.

### Describe the feature you want

The firmware could allow the user to enter different values for speed, acceleration and jerk, depending on the axis. When computing a move, Marlin could calculate the speed and acceleration vectors along each axis, and compare them to user defined limits, then use the smaller one.
Of course, this would not improve print quality. But it could increase speed (reduce printing time).
Trivial case, a square : if there's too much ringing along Y axis, we reduce speed and acceleration, but we lose speed along X axis. Solution : we could use two sets of settings.
Non trivial case : a random move. Speed and acceleration vectors could be computed so their projections on the X and Y axis are not greater than the predefined values for these axis. Some simple math, but probably not that easy to implement....

### Additional context

I was thinking of that while looking at results for an embossed text around a vertical conical surface, printed at high speed, and searching the sweet spot.
Anyone can also see this on a simple calibration cube.

Would be such a feature be usefull, and significantely reduce printing time ? No idea. Depends a lot of the model size and it's form factor... At least, it could give in constistent result along any surface/curve.

iZz0s, 2021-05-16:
you can limit max acceleration for each axis

example:
print acc: 1000
max acc x: 1000
max acc y: 500

this will limit y to 500, even if print acc is 1000.

yet-another-average-joe, 2021-05-16:
Oops...

You are right. I've been using WinMerge for so long while converting Configuration.h from one version to another that I didn't even remember ; just remembered about DEFAULT_MAX_FEEDRATE and DEFAULT_MAX_ACCELERATION.

I've been thinking of independant settings while looking at prints sliced with Cura, and Cura does not make any difference. These settings are maxed up in Marlin, I control speed, acceleration and jerk from the slicer.

This is a feature request for a slicer plugin ! Arc Welder could be a good candidate.

This FQ is irrelevant, I close it right now !

Sorry for the inconvenience !

closed 2021-05-16 by yet-another-average-joe