MarlinFirmware/Marlin #21928: [BUG] mks_robin_nano35_stm32 not generating Robin_nano35.bin

th0m4sek, 2021-05-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

compilation finishes without problems but there is no Robin_nano35.bin in build catalog

### Version of Marlin Firmware

bugfix-2.0.x

ellensp, 2021-05-16:
buildroot/share/PlatformIO/scripts/mks_encrypt.py is saving the file as build.firmware, vs  using board_build.firmware = Robin_nano35.bin

ellensp, 2021-05-16:
In buildroot/share/PlatformIO/scripts/mks_encrypt.py  line 20 
change	`marlin.encrypt_mks(source, target, env, "build.firmware")`
to		`marlin.encrypt_mks(source, target, env, board.get("build.firmware"))`

this works, but its 3am so ill have to do a PR later

commit c3a65eef (2021-05-17): Fix envs using mks_encrypt.py (#21933)

Fix #21928
commit 5d7c72db (2021-05-17): Fix envs using mks_encrypt.py (#21933)

Fix #21928
closed 2021-05-17 by thinkyhead