MarlinFirmware/Marlin #21929: [BUG] EEPROM_CHITCHAT No Feedback?

iZz0s, 2021-05-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

No real big problem, maybe a small bug?

I've found out that my printers with current marlins don't give Feedback, as example, on a simple M500. It just writes an "ok".

Normally, if i remember correctly, it will write something like "EEPROM saved (xxx bytes)".

ofcourse EEPROM_CHITCHAT is activated in firmware.

Send:18:34:41.770: N17 M500
Recv:18:34:42.157: ok N17 P31 B3
Recv:18:34:46.155: wait (4)
Send:18:34:46.212: N18 M500
Recv:18:34:46.598: ok N18 P31 B3
Recv:18:34:47.597: wait

### Expected behavior

EEPROM_CHITCHAT should confirm eeprom changes with a message with saved bytes etc.

### Actual behavior

EEPROM_CHITCHAT only writes in console.

### Steps to Reproduce

M500

### Version of Marlin Firmware

newest bugfix

### Printer model

sv01

### Electronics

skr mini e3 2.0

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

viper93458, 2021-05-16:
I’ve noticed that too for a while using octoprint. I thought it was just something in my config. :)

ellensp, 2021-05-17:
This PR fixes it

ellensp, 2021-05-18:
Merged, closing

commit 1e189506 (2021-05-18): Fix EEPROM_CHITCHAT (#21934)

Fix #21929
commit dbb8f3db (2021-05-18): Fix EEPROM_CHITCHAT (#21934)

Fix #21929
closed 2021-05-18 by ellensp