MarlinFirmware/Marlin #21940: [BUG] printer stops, heaters go off and LCD freeze

Balckandy, 2021-05-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When i print form Sd card, in some random point printer stops heaters go off and LCD freeze. 

### Version of Marlin Firmware

 Marlin 2.0.8.1

### Printer model

Geetech A10M

### Electronics

gt2560 v4.0

### Add-ons

BL TOUCH

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6498242/Marlin.zip)

ellensp, 2021-05-18:
reformat (long not quick) your SD card, it is probably corrupt.

Vertabreak, 2021-05-18:
looking over the config. 
//#define E1_DRIVER_TYPE A4988
should be uncommented and the feed rates being uncomfortably high, more then double what i use.
i don't see anything that would cause the noted behavior in the configs nor have i experienced it on my A10M.
i would have to agree there must be a issue with the sdcard it self.

Balckandy, 2021-05-18:
tnks for ur quickly reply, I will try with the SD reformat, and correct the FW.

Balckandy, 2021-05-20:
hi, the problem was solve, aparently was the SD card damage, tnks for ur reply.

closed 2021-05-20 by Balckandy