MarlinFirmware/Marlin #21941: Compile error: U8glib.h: no such file or folder

JozsefKatus, 2021-05-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I download the 2.0.8.1 (same situation with 2.0.8) and I compile it without any changes, than works fine. But if I set (even only) in the LCD-section:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
I get this error message:
Z:\nyomtato upgrade\Firmware\RAMPS\Marlin-2.0.8.1\Marlin\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp:65:10: fatal error: U8glib.h: No such file or directory
 #include <U8glib.h>
          ^~~~~~~~~~
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board Arduino Mega or Mega 2560. (Error compiling...)
I tried followings: 
-PlatformIO (same behavior)
-Update: Python, ArduinoIDE, VSCode, PlatformIO
-Under Win7, Win10 

### Bug Timeline

New issue, started with 2.0.8

### Expected behavior

Successful compile :)

### Actual behavior

See description.

### Steps to Reproduce

1. Download Marlin 2.0.8.x
2. set #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
3. Compile/Upload

### Version of Marlin Firmware

2.0.8.1

### Printer model

i3-like custom

### Electronics

Stock Mega2560 w. RAMPS 1.6

### Your Slicer

Slic3r

### Host Software

Repetier Host

ellensp, 2021-05-18:
I get no such error. Under platformio when enabling REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
I suspect there is an issue with your platfomio.  In your home directory delete .platformio folder. for force it to re download build environments and libraries

JozsefKatus, 2021-05-18:
The first configs are the real files for the printer. The second are the test files wit only this setting
[Marlin2.8.0config.zip](https://github.com/MarlinFirmware/Marlin/files/6499263/Marlin2.8.0config.zip)
[Marlin8.0.8.1_test_config.zip](https://github.com/MarlinFirmware/Marlin/files/6499272/Marlin8.0.8.1_test_config.zip)

ellensp, 2021-05-18:
also remove .pio folder in marlin and try again

ellensp, 2021-05-18:
Your test config compiles fine.

JozsefKatus, 2021-05-18:
I didn´t find any folder with the names: .platformio (in Home) or .pio (in Marlin). Maybe it is a similar user mistake, I´m not really familiar with PlatformIO.
I tried on another computer: fresh installed Win10 and anything other. Now I get another error message, see attachment. The same with ArduinoIDE.
Many thanks! 
[Arduino_error.txt.txt](https://github.com/MarlinFirmware/Marlin/files/6499566/Arduino_error.txt.txt)
[image]

ellensp, 2021-05-18:
. directories are hidden directories
you need to tell the os to display them to you

closed 2021-05-18 by ellensp