MarlinFirmware/Marlin #21943: After upgrading to 2.0.801 Y homming switch do not work.

tubaistvan, 2021-05-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After updating from "bugfix-2.0.x" date "2021-03-14" to "2.0.8.1" date "2021-05-15" Y homming switch do not work.

### Version of Marlin Firmware

MARLIN 2.0.801

### Printer model

E3D Tool Changer

### Electronics

BIG TREE TECH GTR V1.0

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

Changing back to "bugfix-2.0.x" date "2021-03-14" works ok.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6500569/Configuration.zip)

ellensp, 2021-05-18:
There are two build environment, which did you use or did you try both BIGTREE_GTR_V1_0 and BIGTREE_GTR_V1_0_usb_flash_drive ?

tubaistvan, 2021-05-18:
Only BIGTREE_GTR_V1_0.

ellensp, 2021-05-19:
Anyone else got a BOARD_BTT_GTR_V1_0 and can test if this is a problem for all users BTT_GTR_V1_0 users

Bauschlumpf, 2021-05-22:
Hello,
I have BTT_GTR_V1_0 and can home all axis separate from Marlin menu (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) and 3.5 TFT via serial 3 with 2.0.8.0 and 2.8.8.1, one extruder, BLTouch and two Z-Axis drivers.

I have an other problem with 2.0.8.1: I can't use a USB connection from repetier server/OctoPrint. V2.0.7.2 works fine. 

[Bauschlumpf-Config.zip](https://github.com/MarlinFirmware/Marlin/files/6526643/Bauschlumpf-Config.zip)

Bauschlumpf, 2021-05-22:
@ tubaistvan: I can't see any stepper definition in your config (from line 746 ...

tubaistvan, 2021-05-24:
Ok, I found the solution.
If use the new definition off Toolchanger servo, the Y endstop do not work.
#define SOL0_PIN                            PH13   // Toolchanger
I switched back to the old line, all right.
#define SERVO1_PIN                          PH13

Bauschlumpf, 2021-05-24:
@tubaistvan: Is it possible to get all your working configuration including platformio.ini? So you can help me to solve my problem too. I find tool changer enormously exciting.

tubaistvan, 2021-05-26:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6545446/Configuration.zip)
I attached the files.
I use Octoprint without a problem.

Bauschlumpf, 2021-05-27:
> I attached the files.
> I use Octoprint without a problem.

Thank you!!!

thisiskeithb, 2021-11-29:
> Ok, I found the solution.
If use the new definition off Toolchanger servo, the Y endstop do not work.
#define SOL0_PIN PH13 // Toolchanger
I switched back to the old line, all right.
#define SERVO1_PIN PH13

`PH13` is not a valid pin on the GTR, so I’m not sure how you got that working.

https://github.com/MarlinFirmware/Marlin/pull/17239/ changed some related pins, but that doesn’t match what you’re reporting here.

Is this still an issue with the latest Marlin `2.0.9.2`?

closed 2022-04-12 by github-actions[bot]