MarlinFirmware/Marlin #21945: [BUG] (bug summary) Marlin 2.0.8 and 2.0.8.1 won't build and gives an error for UNIPANEL in the configuration_adv.h file.

hopeinformer, 2021-05-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have searched for a solution but could not locate a fix that works. I only noticed this error after the last 2 updates. Granite, I am just using my previous Configuration.h and Configuration_adv.h from the Marlin 2.0.7 and changing the values for my printer setup.

Here is the exact error from the log: 
In file included from buildroot/share/PlatformIO/scripts/common-dependencies.h:48:
buildroot/share/PlatformIO/scripts/../../../../Marlin/Configuration_adv.h:1139:19: error: #if with no expression
 1139 |   #if IS_ULTIPANEL
      |                   ^
CalledProcessError: Command 'arm-none-eabi-g++ -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "/home/bowlcut3x/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 175:
    env.SConscript(item, exports="env")
  File "/home/bowlcut3x/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/bowlcut3x/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/bowlcut3x/Dropbox/3D Printer/Marlin 2.0x/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 280:
    apply_features_config()
  File "/home/bowlcut3x/Dropbox/3D Printer/Marlin 2.0x/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 112:
    if not env.MarlinFeatureIsEnabled(feature):
  File "/home/bowlcut3x/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/home/bowlcut3x/Dropbox/3D Printer/Marlin 2.0x/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 256:
    load_marlin_features()
  File "/home/bowlcut3x/Dropbox/3D Printer/Marlin 2.0x/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 244:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "/usr/lib/python3.8/subprocess.py", line 411:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 512:
    raise CalledProcessError(retcode, process.args,

### Bug Timeline

This only happens with the 2.0.8 and 2.0.8.1

### Steps to Reproduce

1. Load Marlin folder into Visual Studio Code
2. click the build button, or if "Auto Build Marlin" plugin is working click build. This is a hit or miss plugin.
3. The build will fail with the above error. 
4. I replace the 2 files "Configuration.h and Configuration_adv.h" with the previous release from Marlin 2.0.7 and there's no issues with building the firmware.

### Version of Marlin Firmware

2.0.8

### Printer model

CORE-XY

### Additional information & file uploads

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/6503520/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/6503521/Configuration_adv.h.txt)

thisiskeithb, 2021-05-18:
Don’t recycle your configs. Options and names change, so you’ll need to compare your old configs with the new ones and transfer your settings over.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-05-18 by thisiskeithb

hopeinformer, 2021-05-18:
You're brilliant. Thank you. I just downloaded the newest files from Marlin and didn't realize the updates were suppose to come with all the other files and directories. Thank you for your help. You want me to delete this thread?