MarlinFirmware/Marlin #21946: [BUG] STM32F103RC_btt_USB_stm32 does not compile for SKR MINI E3 V2

tweinschenk, 2021-05-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I just switched from 2.0.8 to 2.0.8.1 and the new environment from STM32F103RC_btt_USB to STM32F103RC_btt_USB_stm32 but that does not compile. The STM32F103RC_btt_USB and the STM32F103RC_btt_stm32 do work but not the stm32 with USB. I added the build log and my config.

[build_log.txt](https://github.com/MarlinFirmware/Marlin/files/6504374/build_log.txt)

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6504380/configuration.zip)

### Bug Timeline

New in 2.0.8.1

### Expected behavior

The build should work :)

### Actual behavior

The build failed. :(

### Steps to Reproduce

1. change default_envs in platformio.ini to STM32F103RC_btt_USB_stm32
2. try to build

### Version of Marlin Firmware

2.0.8.1 / also in the bugfix-2.0.x branch

### Printer model

Ender 3 Pro

### Electronics

SKR MINI E3 V2 and BLTOUCH

### Your Slicer

Cura

### Host Software

OctoPrint

iZz0s, 2021-05-18:
can confirm that bug,

STM32F103RC_btt_USB_stm32 failed but STM32F103RC_btt_stm32 works.

by the way, can someone explain me the difference between those new "stm32" versions and the old like "STM32F103RC_btt" ?

and when its better tp use the usb ot the non-usb version? i don't understand the real difference, as i can both versions via repetier server.

thisiskeithb, 2021-05-18:
> by the way, can someone explain me the difference between those new "stm32" versions and the old like "STM32F103RC_btt" ?

The `*_stm32` environments compile these boards under the STM32 HAL which will replace STM32F1 HAL eventually:
https://github.com/MarlinFirmware/Marlin/blob/ad8c16177e825c20c8868ac2d440762451d29c60/ini/stm32f1.ini#L146-L150

rhapsodyv, 2021-05-19:
Yes, it doesn't work yet. We are waiting a reply from stm32duino. 

Soon it will be fixed.

rhapsodyv, 2021-06-04:
Can you test bugfix again? A fix was merged.

tweinschenk, 2021-06-04:
I think the STM32F103RC_btt_USB is now the STM32 HAL - did I got this right?
So I tried to build STM32F103RC_btt_USB in the bugfix branch but it still fails.

ellensp, 2021-06-04:
yes STM32F103RC_btt_USB_stm32 became STM32F103RC_btt_USB 

I just tried your configs in current marlin bugfix and it compiled fine in STM32F103RC_btt_USB

```
Environment          Status    Duration
-------------------  --------  ------------
STM32F103RC_btt_USB  SUCCESS   00:00:24.105

```
For completeness I also tried the orrginal build environment now called STM32F103RC_btt_USB_maple

```
Environment                Status    Duration
-------------------------  --------  ------------
STM32F103RC_btt_USB_maple  SUCCESS   00:00:14.984

```
Make clean and try again.

tweinschenk, 2021-06-04:
Oke, I tried again, it worked now :) I just deleted the .pio and the .vscode Folder now - it seems the clean task was not enough.

closed 2021-06-04 by thisiskeithb