MarlinFirmware/Marlin #21947: [FR] Supporting true Core-XYZ mode

kyrreaa, 2021-05-18:
### Is your feature request related to a problem? Please describe.

no

### Are you looking for hardware support?

no

### Describe the feature you want

Marlin supports all kinds of Core-?? setups with both normal and reversed pair of axis involved.
It does however not appear to support a true Core-XYZ machine which I have created.
I am currently testing it out longterm (stresstesting) and so far it is proving quite reliable, but I am running a modified Repetier Firmware to do so. It needs a full kinematic for X, Y, Z -> A, B, C, D motor transform to move in all 3 axis.

Motor naming: (As seen from above)
Back
D C
A B
Front

Motor rotation is seen from shaft side of motor; face on.
Contributions from Motors ABCD on motion XYZ: (+N denotes positive axis motion contribution from clockwise rotation of motor N.)
X = +A +B -C -D
Y = +A -B -C +D
Z = +A -B +C -D

The ABCD data would then be:
A = X + Y + Z
B = X - Y - Z
C = -X - Y + Z
D = -X + Y - Z

It thus takes cartesian input and generates steps on 4 motors. None of which can just be paralleled...

### Additional context

Videos of machine running:
https://youtu.be/9iePaK2-MS4
https://youtu.be/Dk_4NDFEFgI

kyrreaa, 2021-05-19:
Looking into the code I see that in order to handle this the code must be expanded to handle axis A, B, C, D as it is only enabled for a 2:2+1:1 transform now. The XYZ is overcomplicated here and overconstrained (meaning there are step combos that are illegal like stepping only a single motor) but when driven correctly it seems to balance out any binding etc well.

I guess AxisEnum needs to be expanded and renumbered before any other extensions are made.
Planners get_axis_position_mm would need a third core axis check as well.
The more I look at it the more work I see.

DerAndere1, 2021-06-07:
up to 6 axes are possible in bugfix-2.0.x since some days.

kyrreaa, 2021-06-07:
Wohooo! This will be fun! I'll see if my transforms from Repetier can be transposed into Marlin. What was done there was create a sort of transform matrix allowing user to define the way the axis convert thus being fully generic.

DerAndere1, 2021-06-08:
Have a look at the HANGPRINTER kinematics (https://github.com/MarlinFirmware/Marlin/pull/9180) that was present in Marlin 1.1.9 to see how to implement inverse kinematics for a machine with 4 kinematic axes that can be controlled by sending cartesian positions as G-code (Syntax: G1 Xxxx Yyyy Zzzzz Eeeee Fffff) .