MarlinFirmware/Marlin #21948: [BUG] I can´t compile Marlin with LCD error

javigp540, 2021-05-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

I am trying to install the latest version of Marlin on my printer and keep showing the following error each time I try to compile.

Arduino:1.8.15 (Windows 10), Tarjeta:"Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\marlinui_DOGM.h:31:0,

                 from C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\marlinui.cpp:126:

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:37:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:40:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_4X_HAL::init(pin_t, pin_t, pin_t, pin_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:53:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7920_128x64_HAL_4x_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset); // a0 = U8G_PIN_NONE

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_4X_HAL::init(pin_t, pin_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:56:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7920_128x64_HAL_4x_hw_spi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset); // a0 = U8G_PIN_NONE

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_RRD::init(pin_t, pin_t, pin_t, pin_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:71:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7920_128x64_rrd_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset);   // a0 = U8G_PIN_NONE

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SH1106_128X64_2X_I2C_2_WIRE::init(uint8_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:81:59: error: 'init' is not a member of 'U8GLIB'

   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_sh1106_128x64_2x_i2c_2_wire, options); }

                                                           ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1306_128X64_2X_I2C_2_WIRE::init(uint8_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:90:59: error: 'init' is not a member of 'U8GLIB'

   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_ssd1306_128x64_2x_i2c_2_wire, options); }

                                                           ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_TFT_320X240_UPSCALE_FROM_128X64::init(uint8_t, uint8_t, uint8_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:103:77: error: 'init' is not a member of 'U8GLIB'

   void init(uint8_t cs, uint8_t rs, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_tft_320x240_upscale_from_128x64, cs, rs, reset); }

                                                                             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_MINI12864_2X_HAL::init(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:115:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_uc1701_mini12864_HAL_2x_sw_spi, sck, mosi, cs, a0, reset);

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_MINI12864_2X_HAL::init(uint8_t, uint8_t, uint8_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:118:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_uc1701_mini12864_HAL_2x_hw_spi, cs, a0, reset);

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:131:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_ssd1309_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t)':

C:\Users\Garcia\Desktop\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:134:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_ssd1309_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

exit status 1

Error compilando para la tarjeta Arduino Mega or Mega 2560.

I did try an older version from Marlin but keep showing the same error.

### Expected behavior

I expect to compile with the right LCD drive

### Version of Marlin Firmware

Marlin 2.0x

### Printer model

Creality CR-10 S5

### Electronics

Stock electronics with BL Touch

ellensp, 2021-05-19:
Arduino IDE uses wrong U8G library
Use platformio https://marlinfw.org/docs/basics/install_platformio_vscode.html

borjarodi, 2021-05-19:
Hello, I also have a similar error and I don't know how to solve it. In my case, Ender 5 Plus trying to compile the latest version of Marlin and I also have error with LCD.

[image]

ellensp, 2021-05-19:
@borjarodi  don't hijack issues.
your issues is clearly something different and an already ongoing issue https://github.com/MarlinFirmware/Marlin/issues/21576

borjarodi, 2021-05-19:
Hi, it's my first time compiling Marlin and I'm getting this same error. Attached capture of the error that VS shows me. I would appreciate help as I am a beginner
[image]

closed 2021-05-19 by ellensp

Angel996, 2021-05-22:
I downloaded the Marlin version of the U8glib _(U8glib-HAL-0.4.2.zip)_, how do I install it into Arduino IDE? Replacing the old version's files inside the _src_ folder does not work, I'm getting the same compile error. I tried to install the library using the Arduino IDE _Install from zip.._. feature, but I'm getting the "Specified folder/zip file does not contain a valid library" error msg.

Enzo-Enz, 2021-06-18:
Hi @Angel996 
Im tried to add U8glib-HAL-master.zip to arduino IDE but I faced with this error:

Arduino: 1.8.15 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Specified folder/zip file does not contain a valid library

Specified folder/zip file does not contain a valid library

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

could you please help me?

thisiskeithb, 2021-06-18:
@Enzo-Enz: This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server