MarlinFirmware/Marlin #21951: [BUG] Z-offset doesn't work

StefanoChiodino, 2021-05-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

See https://github.com/stefanochiodino/marlin/tree/2.0.8.1-stefano-s-ender-3-v2 for my configuration.

Using the 2.0.1 firmware built by Creality I could adjust the Z-offset to get the right amount, even mid-print. Now my extruder is hovering 1 cm over the bed, and weather I set it to -10 or +10 nothing ever changes.

### Bug Timeline

Wasn't there in 2.0.1 built by Creality

### Expected behavior

I change the Z-offset mid-print to get the right one, the extruder moves accordingly, and this is saved.

### Actual behavior

The Z-offset never changes.

### Steps to Reproduce

Start a print. Go to tune, then change Z-offset. The extrude doesn't move closer or further from the bed.

### Version of Marlin Firmware

2.0.8.1

### Printer model

Creality Ender 3 V2

### Electronics

Board 4.2.2

### Your Slicer

Cura

### Host Software

SD Card (headless)

StefanoChiodino, 2021-05-20:
Update: enabling `BABYSTEP_ZPROBE_OFFSET` fixes it, but I don't think it should be tied into babystepping. Before I couldn't change it no matter what!

legend069, 2021-05-26:
try adjusting BABYSTEP_MULTIPLICATOR_Z

closed 2021-08-09 by github-actions[bot]