MarlinFirmware/Marlin #21952: [BUG] PREVENT_COLD_EXTRUSION build error

jdekarske, 2021-05-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to use the extruder axis as a fourth axis in a robot I'm building. (Bonus: if there is a better way to accomplish turning an extruder axis into an arbitrary axis please let me know)

When commenting
``
// #define PREVENT_COLD_EXTRUSION
``
I receive:
``
Marlin/src/feature/pause.cpp: In function 'bool ensure_safe_temperature(bool, PauseMode)':
Marlin/src/feature/pause.cpp:136:51: error: 'class Temperature' has no member named 'extrude_min_temp'
     thermalManager.setTargetHotend(thermalManager.extrude_min_temp, active_extruder);
                                                   ^~~~~~~~~~~~~~~~
*** [.pio/build/STM32F103RC_btt/src/src/feature/pause.cpp.o] Error 1
``
Looks like the macro prevents the attribute from being assigned in `temperature.cpp`

### Bug Timeline

just noticed

### Steps to Reproduce

1. Comment `#define PREVENT_COLD_EXTRUSION` in Configuration.h
2. Build

### Version of Marlin Firmware

2.0.7

### Printer model

NA

### Electronics

BTT SKR MINI E3 V2

jdekarske, 2021-05-21:
I needed to set `#define EXTRUDE_MINTEMP -100` since an unattached temperature sensor reads -15 I guess.

thisiskeithb, 2021-05-21:
You’re trying to force Marlin to do something that it can’t do yet.

Support for more axes are coming: https://github.com/MarlinFirmware/Marlin/pull/19112

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-05-21 by thisiskeithb

jdekarske, 2021-05-21:
Hi Kieth, I'll look in to contributing there, thank you.
Despite this, you still can't turn off cold extrusion without receiving an error.

thinkyhead, 2021-05-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

jdekarske, 2021-05-21:
Ah yes, after trying with a fresh bugfix branch, the issue seems resolved. Sorry for any trouble.