MarlinFirmware/Marlin #21954: [BUG] (bug summary)

cappuccino02, 2021-05-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The second extruder in double X-axis mode cannot work normally，DUPLICATION，MIRRORED

### Version of Marlin Firmware

2.0.8.1

### Electronics

BIGTREETECH SKR PRO 1.2

### Your Slicer

Cura

ellensp, 2021-05-21:
Please attach your configuration files.
Other have it working...

It could be that you have DISABLE_INACTIVE_EXTRUDER enabled, in which case this is a duplicate of https://github.com/MarlinFirmware/Marlin/issues/21857

ellensp, 2021-05-25:
No configs, cannot do anything with this, but close it.

closed 2021-05-25 by ellensp