MarlinFirmware/Marlin #21960: [BUG] MKS sGen L v2.0 does not home with sensorless homing in v2.0.8.1 vs 2.0.7.2

ebraiman, 2021-05-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Sensorless homing fails in Marlin 2.0.8.1 and works in Marlin v2.0.71

[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/6525501/Marlin-2.0.x.zip)
[Uploading Marlin-2.0.7.2.zip…]()

### Bug Timeline

New Bug

### Expected behavior

Sensorless homing would work like Marlin v2.0.7.2

### Actual behavior

Sensorless homing works on both firmware configurations for MKS sGen L v2.0

### Steps to Reproduce

Step 1. Download Marlin firmware from www.marlinfw.org
Step 2. Extract firmware
Step 3. Open in VScode
Step 4. Edit platformio.ini for default environment to be from MEGA2560 to LPC1769
Step 5. Change mother type for MKS sGen L v2 (BOARD_MKS_SGEN_L_V2) 
Step 6. #define SERIAL_PORT -1
Step 7. #define X_DRIVER_TYPE  TMC2209
Step 8. Enable EEPROM
Step 9. On configuration_adv.h Enable TMC debug(s)
Step 10. Enable Sensorless homing and set selectivity to 125 for x axis

Follow normal process for configuration jumpers for MKS sGen L v2 for UART on x-Axis and Sensorless homing jumper for x(min)

### Version of Marlin Firmware

2.0.8.1

### Printer model

Board tested on desk.

### Electronics

MKS sGen L v2.0 (v3)

### Add-ons

Makerbase TMC 2209 V2.0

### Your Slicer

Slic3r

### Host Software

Pronterface

### Additional information & file uploads

Found a German bug addressing simlar issue on ReopRap Forum.
https://reprap.org/forum/read.php?247,884020,884074#msg-884074
https://github.com/makerbase-mks/MKS-SGEN_L-V2

YouTube Stream of Issue
https://www.youtube.com/watch?v=vxbVvrDzB_A

thisiskeithb, 2021-05-22:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ebraiman, 2021-05-22:
This appears to be not a bug as M502 appears to reset EEPROM Hardcoded Default Settings.

closed 2021-05-22 by ebraiman