MarlinFirmware/Marlin #21965: [BUG] X and Y axis is overwritten after Auto Homing or Bilinear Leveling

Allaboutzurich, 2021-05-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

**Device & Firmware:** 
Ender 3 v2 with 4.2.7 Mainboard (tried also on 4.2.2), BLTouch Smart 3.1 and Firmware bugfix-2.0.x and Configuration 2.0.8.1.

### Bug Timeline

With all Firmware witch enable the BLTocu it happens

### Actual behavior

When I click "Auto Home" it homes correctly in the middle of the bed and stays there after homing. The problem is, it sets the coordinates to 0/0/10 (XYZ). When I start a print or level sequence from there, it crashes in X or Y axis or sometimes also both. 

When I disable the steppers, put the bed and nozzle to real X 0 and Y0 and then start the bilinear leveling sequence, it goes like this: 

1. Pos. Bottom Left
2. Pos. Bottom Middle, 
3. Pos. Bottom Right
4. Pos. Middle Right
5. Pos. Top Right 
6. Pos. Out of Bed and crash into X max axis

### Steps to Reproduce

1. Click Auto Home (wait until it homed correctly)
2. Click Auto Home again or Click on Auto Bed Leveling

### Version of Marlin Firmware

Bugdix 2.0.X with Config 2.0.8.1

### Printer model

Creality Ender 3 V2

### Electronics

4.2.7

### Add-ons

BLTouch Smart 3.1

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6527013/Marlin.zip)

thisiskeithb, 2021-05-22:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Allaboutzurich, 2021-05-22:
Alright, I'll do that this weekend. - Thanks for your help.

Allaboutzurich, 2021-05-24:
Hi @thisiskeithb, I just replicated the procedure and also recorded the log in a TXT and Video. 

Attached you can find the TXT with the Log and under the following link the video recording of it.
https://youtu.be/05SSVh4SJ74

[Leveling.txt](https://github.com/MarlinFirmware/Marlin/files/6534466/Leveling.txt)

I really hope that helps to find the problem and find a solution. Off course I will show my gratitude for the work with a TIP. :)

Allaboutzurich, 2021-05-24:
I just looked trough the TXT Logfile and the first "unusual" behaviour started at line 385:

Recv: >  X136.00 Y116.00 Z5.09
Recv: <<< do_blocking_move_to  X136.00 Y116.00 Z5.09
Recv:   current_position= X136.00 Y116.00 Z5.09 : Probe::set_deployed

The code sais correctly that it should move to X136 but instead it stayed on position X220.
The weird thing is, that the system thinks it is on position X136, what is obviously wrong. 

The X stepper just stayed as it is instead of moving. 

My findings so far.

ibrushd1, 2022-02-06:
has anyone ever solved this issue? I have the same exact problem but mines with the cr touch using the 8 bit v2.1 main board which is supposed to be impossible but after 2 weeks I am successful minus this stupid auto home in the y axis issue that was never there and goes away if I install the creality bl touch firmware hex file which is insufficient for me with needing custom settings not supported in that firmware! I am using bugfix 2.0 x and configuration example file for cr10s. this y auto home bug is my final issue and itr affects the back row of leveling when the y axis crashes into the y limit switch and the firmware still wants to go back further to level the last row...other than that I have successfully accomplished the supposed impossible by getting the new cr touch to work on an 8 bit v2.1 board and complete a print successfully and beautifully! If this auto home issue ever gets resolved i will be 100% successful at that point.

tombrazier, 2022-02-23:
This looks like a duplicate of bug #21965.

tombrazier, 2022-02-24:
@ibrushd1 I might be able to work with you to find this bug. Some questions:
- Is your printer's behaviour exactly the same as in @Allaboutzurich's video? i.e. during leveling it zigs but does not zag.
- Have you tried @Hyenadae's solution of limiting motor speed and acceleration (presumably via M203 and M201)?
- Are you using the latest bugfix-2.0.x code?
- Can you zip up your config files plus any local changes you have and upload them?
- Just to confirm your exact hardware: I think you are saying you have a Creality CR-10 with a v2.1 mainboard. Is this right? I cannot find any info on the v2.1 mainboard. Any idea where I can find a schematic?

tombrazier, 2022-02-24:
@Allaboutzurich are you still watching this bug? If so you might be able to answer some of my questions @ibrushd1 above.

Allaboutzurich, 2022-02-25:
Hey @tombrazier, yes I'm still watching it, but since no one had a clue I decided to manually level my 3D printer. But if we could find a sollution, it would be awesome. 

I'll try the suggestion this weekend and answer your questions if possible. I'll keep you posted

thinkyhead, 2022-04-10:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

hypov8, 2022-04-20:
I just tried marlin for the first time(release ver) and think I have same issue.
It seems to be from my probe being in a positive +X pos. Causing it to go to an invalid location when X home was at -32. 
This caused tool crashing into oblivion.

The bugfix version seems better but still not fully fixed. After homing X Y then Z it still has invalid XY cords(eg previous homed X-32 Y-5). So after Z home(probe in middle bed XY) if I try move X/Y it will abruptly go back to the original (X-32 Y-5) home position then do my requested move.

The firmware needs to move the head into Z/probe position(eg x100 y100), THEN do the tool change offset. 
Not offset the tool then moving to z/probing location.
If implemented, maybe add it as a define.

#define Z_SAFE_HOMING
#define X_MIN_POS -32
#define NOZZLE_TO_PROBE_OFFSET { 45, 0, -1.0 }

This helped me get around the issue on the release firmware.
```
G28 X
G28 Y
G1 X100 Y100 ; move to centre
G28 Z ; tool didnt crash after this
```

This fails
```
G28 X
G28 Y 
G28 Z ; tool change shifted from X-32(home) into -77… CRASH!!!!
```

Repetier has the same issue with my printer.

tombrazier, 2022-04-20:
@hypov8 Your described behaviour sounds like a different issue from this one. The behaviour this issue refers to is best seen in the video linked at [this comment](https://github.com/MarlinFirmware/Marlin/issues/21965#issuecomment-847271039).

Perhaps raise a new issue including your full config. Or ask for help on the Marlink #support discord channel. It may be just that you have a config issue.

thinkyhead, 2022-04-21:
> @Allaboutzurich : "The code [says] correctly that it should move to X136 but instead it stayed on position X220."
> ```
> Recv: >  X136.00 Y116.00 Z5.09
> Recv: <<< do_blocking_move_to  X136.00 Y116.00 Z5.09
> Recv:   current_position= X136.00 Y116.00 Z5.09 : Probe::set_deployed
> ```

I think the lines you're looking for are these:
```
>>> Probe::probe_at_point  X220.00 Y116.00 Z5.09
...(94.00, 110.00, raise, 0, probe_relative)
  current_position= X220.00 Y116.00 Z5.09 :
>>> do_blocking_move_to  X220.00 Y116.00 Z5.09
>  X136.00 Y116.00 Z5.09
<<< do_blocking_move_to  X136.00 Y116.00 Z5.09
  current_position= X136.00 Y116.00 Z5.09 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X136.00 Y116.00 Z5.09
>>> Probe::probe_down_to_z  X136.00 Y116.00 Z5.09
BLTouch DEPLOY requested
BLTouch Command :10
bltouch.deploy_proc() end
>>> do_blocking_move_to  X136.00 Y116.00 Z5.09
>  X136.00 Y116.00 Z-2.00
<<< do_blocking_move_to  X136.00 Y116.00 Z-2.00
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
  current_position= X136.00 Y116.00 Z0.09 : sync_plan_position
<<< Probe::probe_down_to_z  X136.00 Y116.00 Z0.09
<<< Probe::run_z_probe  X136.00 Y116.00 Z0.09
>>> do_blocking_move_to  X136.00 Y116.00 Z0.09
>  X136.00 Y116.00 Z5.09
<<< do_blocking_move_to  X136.00 Y116.00 Z5.09
<<< Probe::probe_at_point  X136.00 Y116.00 Z5.09
```

The firmware attempts to command the move, so if the axis isn't actually moving that is mysterious. It could be that the move is too fast and triggering an STST error in the stepper driver, but that would only apply to Trinamic drivers, and the STST error is usually caused by unusual direction changes.

tombrazier, 2022-04-21:
I find it puzzling as well. However I think the problem is in Marlin because the missing moves take no time. Watching the video @Allaboutzurich posted [here](https://github.com/MarlinFirmware/Marlin/issues/21965#issuecomment-847271039) here are the timings of when the probes finish (i.e. bltouch goes red):

0:37 1st G29 probe completes at X=10, Y=10.
0:44 2nd G29 probe completes at X=94, Y=10.
0:51 3rd G29 probe completes at X=178, Y=10.
0:58 4th G29 probe completes at X=178, Y=110.
1:04 5th G29 probe completes, supposedly at X=94, Y=110, but in reality no move occurred.
1:10 6th G29 probe completes, supposedly at X=10, Y=110, but in reality no move occurred.
1:17 7th G29 probe completes, supposedly at X=10, Y=210, Y move occurred but X is still wrong.
After this the head attempts to move beyond bounds.

Notice the total time for each scan cycle is 7 seconds where the moves occurred and 6 seconds where they did not. Each move takes 1 second as close as I can tell. Another way to look at this is to observe that the bltouch remains red until the horizontal move has finished and there is a distinct time between the Z raise and the light going off after the horizontal move. Except where the horizontal move does not occur the red light goes off immediately when the Z raise completes.

If Marlin really were sending out pulses for the missing horizontal moves, they should take about 1s.

tombrazier, 2022-04-21:
Since this is a nice repeatable bug it should be possible to track it down. But we need someone with a machine that exhibits the fault to do some guided experimentation. One of my first suggestions will be to enable the commented debug code in Planner::buffer_segment(). I also have in mind to add a debug serial output reporting step counts.

@Allaboutzurich just a reminder about my questions above. And to add to that do you fancy doing this debugging work?

Allaboutzurich, 2022-04-21:
Hi @tombrazier, I totally forgot. I'm really sorry. But I'm really glad to see that there is some movement with my case. 

If you can guide me trough the process I should find time this weekend to do bug huntig together with you. 

How should I proceed?

tombrazier, 2022-04-22:
@Allaboutzurich Okay, let's start with latest bugfix branch. Remember to re-enable bilinear leveling if you have moved on to another method in the meantime. Does the same problem still happen with leveling?

Enable the debug code in Planner::buffer_segment() by adding a slash where the code comment tells you to (around line 3020). Enable DEBUG_LEVELING_FEATURE. Use M111 S32 to enable the latter. Upload a log.

Now, let's have some info about your current hardware and config:
- Do you still have the Ender 3 v2 with 4.2.7 Mainboard and BLTouch Smart 3.1?
- What stepper drivers do you have?
- Please upload your the config files you are using.

And then some diagnostics:
- Does the incorrect behaviour happen when starting the leveling via the LCD or issuing G29 or both?
- Does limiting max speed and acceleration make any difference? (i.e. use M203 and M201).
- Does it make any difference using the S parameter to G29? try something like S2100.

That should keep you busy enough for now.

Allaboutzurich, 2022-04-24:
Hey @tombrazier, I just did a debug session.

Here you can find all the informations I have.

**Do you still have the Ender 3 v2 with 4.2.7 Mainboard and BLTouch Smart 3.1?**
Yes, my setup remains the same. Ender 3v2 with 4.2.7 Mainboard, BLTouch Smart 3.1 (I just rechecked everything) meanwhile I also added the originall creality filament sensor (FYI).

**What stepper drivers do you have?**
They have the ID "BJ42D15-26V09" and the LOT No. 201205A
Creality put a sticker on them saying "Creality 42-40)

**Please upload your the config files you are using.**
Here you can find the configs I used and the compiled file: 
[Allaboutzurich - Ender 3v2 -Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/8550667/Allaboutzurich.-.Ender.3v2.-Configuration.Files.zip)

And then some diagnostics:

**Does the incorrect behaviour happen when starting the leveling via the LCD or issuing G29 or both?**
Unfortunately both. (See logs)

**Does limiting max speed and acceleration make any difference? (i.e. use M203 and M201).**
No, it doesn't. (See logs)

Does it make any difference using the S parameter to G29? try something like S2100.
No, it doesn't. (See logs)

Attachements:
Logs : 
[Ender 3v2 - Logs.zip](https://github.com/MarlinFirmware/Marlin/files/8550700/Ender.3v2.-.Logs.zip)

**COMMENTS & THOUGHTS** 
AUTO-BILINEAR_LEVELING: The system says that bilinear leveling isn't enabled, which is strange, because as you can see in the config file, I have uncommented it. (See also log Ender 3v2 - G28 G28.txt - LINE 97)
MANUAL LEVELING: When I send the nozzle after G28 with G1 to X0 Y0, and then move to X136 and then to X222 and then send G28, the nozzle doesn't move to X147, Y115, but to X222 Y115 (but on the lcd it says X147 Y115, so the machine thinks it is right.
G28 after G28: The machine shows a really strange behaviour, If I send a G28 command after another one. (See  log Ender 3v2 - G28 G28.txt)

Thanks again for your help and time with my issue. I really appreciate it!

tombrazier, 2022-04-25:
Fab. I'll look at that. Quick response about leveling being disabled: This is the case before it has a mesh. Can be turned on and off with M420 S when there is a mesh.

But your comment about manual leveling and G28 are very interesting. It looks like I might have been mistaken in thinking @hypov8 has a different issue.

tombrazier, 2022-04-25:
Okay, I have replicated both the G29 and the G28 behaviour you describe! Slight caveat: my hardware is different but I think/hope I am getting the same thing as you.

I get the behaviour when the X min endstop is triggered. Then the moves in G28 and G29 will happily increase X but will not decrease it.

To check we are getting the same thing, can you do
```
G28
G1 X222 Y0
M119
```
The M119 should show the X min endstop is triggered if we are getting the same thing.

Allaboutzurich, 2022-04-25:
I just tried it and as you expected, the X_min is triggered. 

> Send: M119
> Recv: Reporting endstop status
> Recv: x_min: TRIGGERED
> Recv: y_min: TRIGGERED
> Recv: z_min: TRIGGERED
> Recv: z_probe: TRIGGERED
> Recv: filament: open
> Recv: ok

Log-File:
[Ender 3v2 - X_Min_Endstop.zip](https://github.com/MarlinFirmware/Marlin/files/8554572/Ender.3v2.-.X_Min_Endstop.zip)

tombrazier, 2022-04-25:
That will do it. Now we just have to find out why all your endstops are triggered. What does M119 report in other X, Y locations? Does changing Z make any difference?

Allaboutzurich, 2022-04-25:
I just tried several points, here the most relevant one: 

**G28 + M119  (X:147.00 Y:115.00 Z:10.00 E:0.00 Count X:11760 Y:9200 Z:4000):** 
> Send: M119
> Recv: Reporting endstop status
> Recv: x_min: TRIGGERED
> Recv: y_min: open
> Recv: z_min: TRIGGERED
> Recv: z_probe: TRIGGERED
> Recv: filament: open
> Recv: ok

**G1 X0 Y0 (X:0.00 (11760->0) Y:0.00 (9200->0) Z:10.00 (4000->4000) E:0.00 (0->0))**
> Send: M119
> Recv: Reporting endstop status
> Recv: x_min: TRIGGERED
> Recv: y_min: TRIGGERED
> Recv: z_min: TRIGGERED
> Recv: z_probe: TRIGGERED
> Recv: filament: open
> Recv: ok

**G1 X222 Y222 (X:220.00 (0->17600) Y:220.00 (0->17600) Z:10.00 (4000->4000) E:0.00 (0->0)**
> Send: M119
> Recv: Reporting endstop status
> Recv: x_min: TRIGGERED
> Recv: y_min: open
> Recv: z_min: TRIGGERED
> Recv: z_probe: TRIGGERED
> Recv: filament: open
> Recv: ok

**G1 X0 Y110 Z50 (X:0.00 (17600->0) Y:110.00 (17600->8800) Z:50.00 (4000->20000) E:0.00 (0->0))**
> Send: M119
> Recv: Reporting endstop status
> Recv: x_min: TRIGGERED
> Recv: y_min: open
> Recv: z_min: TRIGGERED
> Recv: z_probe: TRIGGERED
> Recv: filament: open
> Recv: ok

**M401 + G1 X110 Y110 Z0 (X:110.00 (0->8800) Y:110.00 (8800->8800) Z:0.00 (20000->0) E:0.00 (0->0))**
> Recv: echo:endstops hit:  P:0.29
> Send: M119
> Recv: Reporting endstop status
> Recv: x_min: TRIGGERED
> Recv: y_min: open
> Recv: z_min: TRIGGERED
> Recv: z_probe: TRIGGERED
> Recv: filament: open
> Recv: ok

Should I manualy check with my multimeter if all endstops are connected to the right endstop switches?

tombrazier, 2022-04-25:
I am not anything like an expert on the BLTouch (or any probe that deploys), but at a guess it might be triggered any time it is not deployed. So hopefully that is not a worry.

Since z_min is unconnected and you have Z_MIN_ENDSTOP_INVERTING = false, I would also expect that to always be triggered. (You might be able to unset USE_ZMIN_PLUG to get rid of this.)

So it looks like X min is behaving badly. How have you ever homed X successfully?

X_MIN_ENDSTOP_INVERTING = false so it is probable that the X endstop is just not connected. Or is your X endstop normally open? In which case you should have X_MIN_ENDSTOP_INVERTING = true.

Anyway, yes, measure the endstop's behaviour with a multimeter. And check it is connected to the right socket on the mainboard.

Allaboutzurich, 2022-04-25:
So I just measured all the endstop connections and they seem fine. 

My guess is, that it is the x_endstop which is defective. I could replace the x_endstop with the z_endstop (since it is no longer used thanks to the Bltouch) and check if it fixes the problem.

But could a defective x_endstop prevent the system from moving in the x direction? In the system it is moving (which you can see in the coordinates changing) but physically something blocks the servos to move in the X- direction. 

Should I give it a try? To answer your question, in the future I always homed X by disabling the stepper motors and pushing the nozzle to the X endstop and the printbed to Y endstop.

tombrazier, 2022-04-25:
Homing and probing call the various do_blocking_move_to() functions. These explicitly stop moving the steppers when endstops are reached. That's how probing and homing work. So, yes, a defective x_endstop could prevent these particular movements.

Yes, have a go with using the z_endstop connector. You'll need to do something like change X_STOP_PIN - probably in Configuration_adv.h. Do you need any guidance on that?

tombrazier, 2022-04-25:
Oh, I might have misunderstood. Yes, let me know how it goes.

Allaboutzurich, 2022-04-25:
We fixed it! 
It was the defective x_endstop_switch which prevented the driver to go back in x- direction. I auto-leveled it now for the first time ever and it finished all 9 leveling steps. 

I permanently replaced the X_endstop_switch and gave it full Auto-Home, Auto-Level go.

I guess from my side we could close the issue, or is there something you want to look into in more detail? 
Is there a way I can show you my gratitude with a small tip for your time, patience and support?

closed 2022-04-25 by thisiskeithb

tombrazier, 2022-04-25:
> Is there a way I can show you my gratitude with a small tip for your time, patience and support?

Enjoy your printer. That's enough for me.

Having said that I know the key Marlin maintainers like thinkyhead and thisiskeithb have github sponsor pages. Supporting them is supporting Marlin.

Allaboutzurich, 2022-04-25:
Alright! Thanks again and I definately will.