MarlinFirmware/Marlin #21966: Azteeg X5 GT with Viki2 LCD TMC Connection Error

blackcatSVK, 2021-05-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I'm using Azteeg X5 GT control board with Viki2 LCD, I have an issue with the marlin firmware.

When I undefined //#define VIKI2 line 2001, everything works smoothly, I use octoprint to connect to the printer and its working perfectly.

However, when I define line 2001 above, the LCD works, scrolling, I can used normal function as expected. However, I got "TMC connection error" and the motors do not work at all.

And I did tried to disconnect the LCD (line 2001 is still defined), and same issue happens.

I have tried Google to find the answer but its not much helped.

Do you guys have any suggestion to fix the issue?

Thank you.

### Bug Timeline

20/04/2021

### Expected behavior

I expect the motor working

### Actual behavior

TMC Connection error

### Version of Marlin Firmware

2.0.7.2

### Printer model

Custom CoreXY

### Electronics

Azteeg X5 GT Board, Viki2 LCD

### Your Slicer

Cura

### Host Software

SD Card (headless)

ellensp, 2021-05-23:
What type of stepper drivers do you have, as it tells us is if is using SPI or serial to talks to them.
Please attach your configuration files, as this answers most questions.

blackcatSVK, 2021-05-23:
I'm using TMC2660, these are my config files.

[Config.docx](https://github.com/MarlinFirmware/Marlin/files/6527735/Config.docx)
[Config_adv.docx](https://github.com/MarlinFirmware/Marlin/files/6527736/Config_adv.docx)

Thank you

ellensp, 2021-05-23:
I cannot read a docx. propitiatory Microsoft only format.

ellensp, 2021-05-23:
Also please send the printer A M122,  ie TMC Debugging 
This will tell us what drivers in particular or all are effected.

ellensp, 2021-05-23:
Looking around I see many with this issue https://reprap.org/forum/read.php?13,863090 
and https://github.com/MarlinFirmware/Marlin/issues/17111  no solutions

ellensp, 2021-05-23:
Do you have TMC_USE_SW_SPI enabled in Configuration_adv.h

blackcatSVK, 2021-05-23:
Hi, this is the zip file of two configs.

M112 reporting "connection error, low voltage" on all X, Y, Z and E.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6527898/Marlin.zip)

blackcatSVK, 2021-05-24:
just another report that if I define Viki2 to use the LCD in marlin, I couldn't connect the printer to octoprint either, I need to revert it with two slashes (sorry English is not my native language so sometime I'm confused whether I should use "comment" or "uncomment" here lol) it if I would like to use octoprint.

blackcatSVK, 2021-05-26:
Hi, please any help

blackcatSVK, 2021-06-08:
> Do you have TMC_USE_SW_SPI enabled in Configuration_adv.h

yes I did, nothing improved.

blackcatSVK, 2021-07-05:
anyone could help, please I'm so desperate about my machine.