MarlinFirmware/Marlin #21967: [COMPILE ERROR] 'init' is not a member of 'U8GLIB'

Angel996, 2021-05-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello!

Trying to compile Marlin 2.0.8.1 for Arduino Mega 2560 with Arduino IDE 1.8.15. Getting this error:

`In file included from C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\marlinui_DOGM.h:31:0,
                 from C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\marlinui.cpp:126:
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:37:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:40:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_4X_HAL::init(pin_t, pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:53:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_st7920_128x64_HAL_4x_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset); // a0 = U8G_PIN_NONE
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_4X_HAL::init(pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:56:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_st7920_128x64_HAL_4x_hw_spi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset); // a0 = U8G_PIN_NONE
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_RRD::init(pin_t, pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:71:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_st7920_128x64_rrd_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset);   // a0 = U8G_PIN_NONE
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SH1106_128X64_2X_I2C_2_WIRE::init(uint8_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:81:59: error: 'init' is not a member of 'U8GLIB'
   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_sh1106_128x64_2x_i2c_2_wire, options); }
                                                           ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1306_128X64_2X_I2C_2_WIRE::init(uint8_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:90:59: error: 'init' is not a member of 'U8GLIB'
   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_ssd1306_128x64_2x_i2c_2_wire, options); }
                                                           ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_TFT_320X240_UPSCALE_FROM_128X64::init(uint8_t, uint8_t, uint8_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:103:77: error: 'init' is not a member of 'U8GLIB'
   void init(uint8_t cs, uint8_t rs, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_tft_320x240_upscale_from_128x64, cs, rs, reset); }
                                                                             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_MINI12864_2X_HAL::init(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:115:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_uc1701_mini12864_HAL_2x_sw_spi, sck, mosi, cs, a0, reset);
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_MINI12864_2X_HAL::init(uint8_t, uint8_t, uint8_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:118:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_uc1701_mini12864_HAL_2x_hw_spi, cs, a0, reset);
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:131:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_ssd1309_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:134:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_ssd1309_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
In file included from C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\marlinui_DOGM.h:31:0,
                 from C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\lcdprint_u8g.cpp:14:
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:37:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:40:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
[... 133 lines cut ...]
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:131:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_ssd1309_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t)':
C:\EXPO\!3D Printer\Prusa i3\Marlin-bugfix-2.0.x\Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:134:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_ssd1309_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
`

There are only two versions of U8Glib available in Marlin, tried either of them, still same error. The U8glib GitHub page says this library has been superceded by U8G2, but Marlin does not seem to be using the newer one.

### Bug Timeline

Version 2.0.8.1 only

### Version of Marlin Firmware

2.0.8.1

### Printer model

prusa i3

thisiskeithb, 2021-05-22:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21948

thisiskeithb, 2021-05-22:
I'd recommend using VSCode & PlatformIO so it automatically selects the correct u8glib library for you. Otherwise, you'll need to tell Arduino IDE to use [Marlin's modified library]([url](https://github.com/MarlinFirmware/U8glib-HAL)).

closed 2021-05-22 by thisiskeithb