MarlinFirmware/Marlin #21970: [BUG] chitu_f103 does not compile

zentaur1ger, 2021-05-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Build fails due to several errors 

### Bug Timeline

NEW 20210523

### Expected behavior

Build should complete successfull

### Steps to Reproduce

1. Open the marlin folder in VS Code
2. Insert BOARD_CHITU3D_V6 in the motherboard definition in configuration.h
3. Insert a custom machine name in configuration.h
4. Define Serial Ports in configuration.h
5. Start build process

Result:
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:30:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:138:1: error: expected ';' before 'void'
 void HAL_init();
 ^~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:30:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:87:3: error: static assertion failed: SERIAL_PORT must be from 1 to 5. You can also use -1 if the board supports Native USB.
   static_assert(false, "SERIAL_PORT must be from 1 to " STRINGIFY(NUM_UARTS) ". You can also use -1 if the board supports Native USB.")
   ^~~~~~~~~~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:30:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:32:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:138:1: error: expected ';' before 'void'
 void HAL_init();
 ^~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:30:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:32:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:87:3: error: static assertion failed: SERIAL_PORT must be from 1 to 5. You can also use -1 if the board supports Native USB.
   static_assert(false, "SERIAL_PORT must be from 1 to " STRINGIFY(NUM_UARTS) ". You can also use -1 if the board supports Native USB.")
   ^~~~~~~~~~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:30:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32F1/MarlinSerial.cpp:25:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:138:1: error: expected ';' before 'void'
 void HAL_init();
 ^~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:30:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32F1/MarlinSerial.cpp:25:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:87:3: error: static assertion failed: SERIAL_PORT must be from 1 to 5. You can also use -1 if the board supports Native USB.
   static_assert(false, "SERIAL_PORT must be from 1 to " STRINGIFY(NUM_UARTS) ". You can also use -1 if the board supports Native USB.")
   ^~~~~~~~~~~~~
*** [.pio/build/chitu_f103/src/src/HAL/STM32F1/HAL.cpp.o] Error 1
*** [.pio/build/chitu_f103/src/src/HAL/STM32F1/HAL_SPI.cpp.o] Error 1
*** [.pio/build/chitu_f103/src/src/HAL/STM32F1/MarlinSerial.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:30:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32F1/SPI.cpp:43:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:138:1: error: expected ';' before 'void'
 void HAL_init();
 ^~~~
[... 98 lines cut ...]

### Version of Marlin Firmware

BUGFIX-2.0.x 20210523

### Printer model

TRONXY XY2PRO

### Electronics

STOCK CHITU3D V6

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[BUILD-LOG.txt](https://github.com/MarlinFirmware/Marlin/files/6527615/BUILD-LOG.txt)
[CONFIGURATION-H_ADV-H.zip](https://github.com/MarlinFirmware/Marlin/files/6527616/CONFIGURATION-H_ADV-H.zip)

ellensp, 2021-05-23:
User cannot read error messages on screen "SERIAL_PORT must be from 1 to 5. You can also use -1 if the board supports Native USB."

closed 2021-05-23 by ellensp

zentaur1ger, 2021-05-23:
> User cannot read error messages on screen "SERIAL_PORT must be from 1 to 5. You can also use -1 if the board supports Native USB."
> 
> Clearly has #define SERIAL_PORT 0 which is not valid

Thanks for the hint, I was probably confused by the rest of the error messages

mrl22, 2021-05-26:
I had the same issue, X5SA Pro.

#define SERIAL_PORT 1
#define SERIAL_PORT_2 3

Fixed.