MarlinFirmware/Marlin #21971: [BUG] Case light dimming via LCD menu not working as expected

Life0fBrian, 2021-05-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Case light dimming is not working via LCD menu. It keeps full brightness between 100% and 10% and then dims when further lowering down to 0%.
Board: RADDS 1.6 on an Arduino Due.
LCD: RADDS-LCD

### Bug Timeline

Bug was already present with Marlin 2.0.7.2 and maybe earlier versions.

### Expected behavior

I would expect a quite step less dimming when decreasing the brightness percentage in the LCD menu having full brightness at 100% and lights off at 0%.

### Actual behavior

Actually it does not change the brightness when decreasing it from 100% down to 11% and then dims on the last 10% down to 0%.
If I verify the behavior using M355 S1 Pxxx with xxx from 0 to 255 it works fine.
As I expect the brightness menu to be a plain conversion of the 256 PWM steps into 100 precentage steps I do not know why it does not work properly.
Even when uncommenting CASE_LIGHT_MAX_PWM and set it to 255 it does not change the bahavior.

### Steps to Reproduce

Uncomment the CASE_LIGHT_ENABLE feature in configuration_adv.h, set the right pin where the LED strip is connected to via a MOSFET and enable the CASE_LIGHT_MENU to control the LED strip via the LCD of my board.
Flash compiled firmware to your board and got the the LCD and enter the Light submenu.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

NoName - CoreXY

### Electronics

RADDS 1.6 + RADDS-LCD

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6527761/Marlin.zip)

Life0fBrian, 2021-06-16:
Upgraded the board to a BTT SKR 2 and Marlin to latest version 2.0.9 and now the dimming works as expected.

thisiskeithb, 2021-06-16:
> Upgraded the board to a BTT SKR 2 and Marlin to latest version 2.0.9 and now the dimming works as expected.

Great!

closed 2021-06-16 by thisiskeithb