MarlinFirmware/Marlin #21973: [Filament Change Retracts on Continue] (Causes layer gap and failed print)

manoman48, 2021-05-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On the Ender 3, the filament change process order is incorrect. The print pauses, the head is parked and the instruction on screen says to insert filament. After manually removing the filament and inserting the new filament, the nozzle heats and the option to purge is given. After purging you can choose continue but at the point the extruder *retracts* ~100mm and so the resumed print fails as there is no filament at the hot end. If you manually push the filament back to the hot end the print can be saved. 

The command to retract needs to be removed following successful purging.

### Bug Timeline

Unknown

### Expected behavior

I expected the print with a primed hot-end to continue following filament change.

### Actual behavior

There was no filament at the hot end as the extruder had retracted it back into the bowden tube following a successful purge.

### Steps to Reproduce

1) From menu choose 'Change Filament'
2) From sub-menu select 'Change Filament'
3) Follow in-screen insctructions

### Version of Marlin Firmware

bugfix-2.0.x 2021-04-14

### Printer model

Creality Ender 3

### Electronics

Creality 4.2.7 - no other upgrades

### Your Slicer

Prusa Slicer

GMagician, 2021-05-23:
This has been probably be fixed by #21670, update your bugfix

manoman48, 2021-05-23:
Thank you. I thought I had updated didn't copy the files across to my project folder. All sorted thank you :)

closed 2021-05-23 by manoman48