MarlinFirmware/Marlin #21974: [BUG] DETECT_BROKEN_ENDSTOP causes halt in IDEX mode

DrumClock, 2021-05-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Opět nefunguje `#define DETECT_BROKEN_ENDSTOP`

Když je koncový doraz X-min zavřený (X1 - parkinkg)
Po odeslání g-kódu G28 se tiskárna zastaví s chybou „parkování selhalo“.

Pokud jsou enstop X-max nebo Y-min uzavřeny, příkaz G28 běží bez problémů.

Tato chyba je pouze v případě, že je povolena funkce  DUAL_X_CARRIAGE !

### Bug Timeline

starý problém který se opět objevil....

### Version of Marlin Firmware

bugfix-2.0.x  (ze dne 23.5.2021)

### Your Slicer

Prusa Slicer

scott-pett, 2021-05-24:
In Dual_X_Carriage mode the printer homes the X2 carriage to the X_Max endstop first.
If this fails, then I see a Homing Error and a Printer HALT

Once X2 is homed, then it homes the X1 carriage to X_MIN.

It had me confused for ages. :-)

Although I'm not sure this is a problem with the DETECT_BROKEN_ENDSTOP function, it is interesting that setting the Y_MIN endstop switch would have an affect on the X/X2 homing process.

DrumClock, 2021-05-24:
Hi @scott-pett 
jedná se o  X_MIN .....k chybě dojde, když je X1 carriage zaparkovaný na dorazu X_MIN  (např. z předešlého tisku)
a je zaslán příkaz G28.

Tato chyba byla dříve odstraněna a nyní je opět zpět a je pouze u IDEX modu.
U jednoho extruderu se funkce  `DETECT_BROKEN_ENDSTOP` provede správně.

Broomer68, 2021-05-30:
When homing X1 (after X2) when X1 already is at X-MIN (and X2 is at X-MAX due to homing), the printer halts.
When X1 is not at X-MIN, or when X2 is not at X-MAX, all goes well.

Workaround: 
#define HOMING_BACKOFF_POST_MM { 3, 3, 2 }  // at least somewhere the switches are freed

It looks to me the detect-broken-endstop checks X-min and X-max where is should only check X-min for X1 and X-max for X2

DrumClock, 2021-05-30:
Ahoj @Broomer68 
funkce `#define HOMING_BACKOFF_POST_MM { 3, 3, 2 }`  na mém IDEX nic neřeší !
Naopak "Printed halted. kill called" je nyní u všech end-stop senzorů (X-MAX, Y-MAX) které
zůstanou před G28 sepnuté....

Broomer68, 2021-05-31:
Check your switches, and the pull-up or pull-down settings.
Not all Pin-configuration are correct for idex-use. (check board and
pin-defenition)

The broken endstops should detect both open-closed or closed-open, and only
reacts on a switch that either remains closed or remains open.
If it is closed, the stepper should move it so it opens, and then move back
home to close again. If it is open, it needs to go directly home. If it
does not open after the move away, or it does not close again after the max
bedsize, it will fail. (this last section can be reversed on all
closed-open states)

The problem with Idex is with the 2 switches concurrently:
X2 homes first, contact x-max gets closed
When x1 needs to home, it already sees x-max, so is not allowed to move and
free up x-min. (Or it moves to x-max, but that will not free x-max.)
My workaround just frees all switches after homing. (unless you need a bit
more space, then increase the numbers given, and check the space between
the extruders, as they may collide at x2-min or x1-max: decrease bedsize or
(soft) -endstop position.

DrumClock, 2021-05-31:
Ten samý problém se řešil 17. února zde https://github.com/MarlinFirmware/Marlin/issues/21107
Nyní je chyba ve funkci ` DETECT_BROKEN_ENDSTOP` pro systém IDEX zpět. Pro mód s jedním extruderem je vše v pořádku.
Není mi jasné proč měnit nastavení když chybu způsobuje jen režim `DUAL_X_CARRIAGE` ?

DrumClock, 2021-07-14:
Because something doesn't work with each new release of bugfix 2.0.x 
(what used to be OK) and nobody solves it, **I decided to switch to FW Klipper.**
**I will no longer pay attention to these issues.**

closed 2021-09-23 by github-actions[bot]