MarlinFirmware/Marlin #21976: SD_Init_Fail [BUG] (bug summary)

singla2602, 2021-05-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have Rerap_Discount_Smart_Controller.
SD card is not working with upadted 2.0.x and 2.0.bugfix firmware.
When I flash old firmware 1.0.x sd card works fine.

### Version of Marlin Firmware

2.0.x & 2.0.bugfix

### Printer model

Folgertech Prusa i3 clone

### Electronics

MKS V1.2

### Additional information & file uploads

SD_Init_Fail

ellensp, 2021-05-24:
Please attach your configuration files.

tig33r, 2021-05-24:
//#define SD_DETECT_STATE HIGH in conf adv

singla2602, 2021-05-25:
thank you problem solved

closed 2021-05-25 by ellensp

tig33r, 2021-05-26:
this is related to https://github.com/MarlinFirmware/Configurations/issues/501