MarlinFirmware/Marlin #21978: [FR] (G34 Auto Level gantry using points off bed for a much better calibration)

andrew4842, 2021-05-24:
### Is your feature request related to a problem? Please describe.

I want to level my gantry x-bar to the frame of my 3d printer but marlin only allows the x-bar to be leveled to the printer bed.
- read why this is a much better way to calibrate under additional context. At the moment G34 is sort of a band aid to make your printer run less terrible when the bed is not level rather then properly leveling the gantry to the frame. That being said with a simple fix and an update in the documentation we can help people properly calibrate their 3d printers.

### Describe the feature you want

I want to level my gantry x-bar to the frame of my 3d printer but Marlin only allows the x-bar to be leveled to the printer bed. 

In my specific example, my bed is 485x585. My probe is my nozzle. My X endstop is 50mm to the left of the start of my bed and I have another 50mm of movement to the right of the bed as well. I want to set my probe points for G34 off the bed so properly level the x-bar to the gantry. 

Why would anyone want to do this?
It's a much much better way to calibrate a 3d printer, let me explain.

So how the feature is intended to work is to level the x-bar to the bed and that is what you want. That being said, if your bed is not level then your x-bar will level to it and subsequently will be crooked which we can all agree can lead to loads of issues. The proper way is to first square the gantry manually, then level the x-bar agents the frame of the 3d printer so everything is nice and square. Then you can move to each corner of the bed and manually move the bed up or down so the bed is level with the entire 3d printer frame. So to level the x-bar with the frame, I normally place 1-2-3 blocks on the aluminum extrusions and have the printer move to one block and probe then to the other until the gantry is level agents the frame. 
Since upgrading to marlin 2.0 I don't want to have to rewrite a custom command to do this again when marlin 2.0 already has G34 built in. 

The only thing I could find was this thread but I don't think they really understood why anyone might want to probe off the bed and therefore no solution was found that could help me. https://github.com/MarlinFirmware/Marlin/issues/16438 

Thanks for your help! I have looked into the code and fully understand why is is not working at all. It makes sense you would have thought that you don't want people to probe off the bed for risk of breaking the 3d printer. That being said it is a much superior way to calibrate a 3d printer and will help people get much better prints. If you disagree please let me know why, I am open to solutions.

andrew4842, 2021-05-25:
I am looking to see how this worked so have gone through the code in G34.cpp, z_stepper_allign.cpp, G34_M422.cpp but for what I am trying to do G34.cpp seems to be what I am looking for, specifically the chunk below. 
[image]

As you can see it homes first which is good, then disables SW endstops. The part I am confused on is where does it go to the XY position because the boxed code seems to only apply to the Z.

Another guess I had was around line 200 in G34_M22
`const float z_probed_height = probe.probe_at_point(z_stepper_align.xy[iprobe], raise_after, 0, true, false);`
This seems to be like "move to z point, probe, and raise up after" 
Also the Z_stepper_align.xy seems to be where those xy coordinates are.
So I went to check back with z_stepper_align.cpp
It seems as the coordinates are stored as `xy_init` .x, .y, or .xy accordingly. I can confirm this because of the LTEST starting around line 59 that seems to get those values.
That being said where do those values come from?
Do they come from Z_STEPPER_ALIGN_XY? I thought this but I cant find where Z_STEPPER_ALIGN_XY comes from then?
I don't really know so help or hints would be wonderful ty!

andrew4842, 2021-06-23:
It's been a month and I still have not been able to work this issue out. Any help would be greatly appreciated!

andrew4842, 2021-06-26:
It seems as though this issue is not important enough to the devs to even get a response but I believe it is very important get an accurate calibration. Below you can see the changes I made to level to points off of the bed and even use M422 to update these points to the Eprom. I hope the devs will eventually come up with a better solution (like a simple uncomment or something) to allow probing off of the bed but until they respond the below changes should work! If you are confused why leveling the X-Bar to the frame is better than leveling it to the bed, I explain it in my first comment last month.

[image]
[image]

thisiskeithb, 2021-06-26:
> In my specific example, my bed is 485x585. My probe is my nozzle. My X endstop is 50mm to the left of the start of my bed and I have another 50mm of movement to the right of the bed as well. I want to set my probe points for G34 off the bed so properly level the x-bar to the gantry.

You simply need to configure your bed size & MIN/MAX settings correctly. Something like this would work for your build:

```cpp
// The size of the printable area
#define X_BED_SIZE 485
#define Y_BED_SIZE 585

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -50
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS (X_BED_SIZE + 50)
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200
```

closed 2021-06-26 by thisiskeithb