MarlinFirmware/Marlin #21981: [FR] (feature summary)

kyon03, 2021-05-25:
### Is your feature request related to a problem? Please describe.

i just noob i wasn't able to port new marlin because of the board not listed thank you very much

### Are you looking for hardware support?

Mingda rock 3 pro using 5 tmc2208

using a MD-301 BOARD_MD_D301      

 D301 V1.0 (STM32F103ZE)

### Describe the feature you want

pls add BOARD_MD_D301 // D301 V1.0 (STM32F103ZE)

DerAndere1, 2021-05-25:
@kyon03 Please edit the issue title. Replace "(feature summary)" by something more descriptive.

ellensp, 2021-05-29:
marlin 2 source from supplier https://www.3dmingda.com/uploads/soft/20210402/Marlin%20source.rar

kyon03, 2021-06-01:
> marlin 2 source from supplier https://www.3dmingda.com/uploads/soft/20210402/Marlin%20source.
2019?

atsju, 2022-03-13:
just to keep track. 
From my research code seems to be based on this commit : 25a7ceabdbac8e60fccc61e31283e7be8b8d5e1a
They rely on a custom board, needing extensibleUI which already makes it a bit difficult.
But they also made some tweeks here and there which wich makes it even more difficult.

My working branch is here https://github.com/atsju/Marlin/tree/JST/21981forTest (for the moment needs to build on MAPLE based environment). I will give it a try to add the UI but will not insist too long if it doesn't work. Manufacturers shall have better maintenance policies. What they did is more like a one shot build for their products, test it, ship it and forget about it.

atsju, 2022-03-13:
@kyon03 @ellensp @lscarneiro @marcellodum @Rock3Pro @doc-jay @Plantgrower @CommanderMoto
I finished the work for supporting your printer based on manufacturers code. 
Please test this version (build with STM32F103ZE_D301_MAPLE):
https://github.com/atsju/Marlin/commit/68d23ab8c6e7655618b9e8a7db32b3f540af6747

It's not finished but I do not own this printer and can not go further without testing.

lscarneiro, 2022-03-13:
> @kyon03 @ellensp @lscarneiro @marcellodum @Rock3Pro @doc-jay @Plantgrower @CommanderMoto I finished the work for supporting your printer based on manufacturers code. Please test this version (build with STM32F103ZE_D301_MAPLE): [atsju@68d23ab](https://github.com/atsju/Marlin/commit/68d23ab8c6e7655618b9e8a7db32b3f540af6747)
> 
> It's not finished but I do not own this printer and can not go further without testing.

I ended up using [@marcellodum's "manufacturer's" fork](https://github.com/marcellodum/marlin-mingda-md-d301) and went with the `TFT_GENERIC` route, which isn't perfect but at least is usable and consistent.

From what I've seen in that fork, @marcellodum found the commit that the manufacturer used and applied all the missing "patches" one by one until `v2.0.9.2` to the "manufacturer's source code", then applied some customizations (which I forked and applied mine on top) and at least for me this made the machine useful and "modern" so to speak.

I think the biggest "problem" is the communication with the touchscreen that is Mingda's branded (which probably is a copycat from another manufacturer). When I tried to build Vanilla Marlin to that chip, I think I got it just fine, but whatever I tried to make the screen work, I failed regardless, to the point that I even thought I bricked the board as I couldn't see the flashing firmware screen anymore, I ended up blindly flashing the manufacturer's `bin` file and got it "working" again. That's when I found the @marcellodum repo.

lscarneiro, 2022-03-13:
> What they did is more like a one shot build for their products, test it, ship it and forget about it.

Yep, totally agree! 

I felt like forgotten by Mingda, you can't even find the printer on their site anymore, and I'm saying that from 2 weeks after I bought the printer, so I definitely got some old stock. I took the risk because I was knowledgeable enough to build marlin for RAMPS boards for quite I while at the time, and the actual hardware was good (5 TMC2208 drivers with double Z motors), so I thought "let's get the good hardware and take care of the software later" but to actually be able to do it I would need to buy a new board and screen, what would make less sense given the increase in the overall cost.

SteveDallas, 2022-08-17:
> I think the biggest "problem" is the communication with the touchscreen that is Mingda's branded (which probably is a copycat from another manufacturer). When I tried to build Vanilla Marlin to that [...]

I have an early Mingda D2 with apparently the first run of the D301 board.  Both the controller and the display are marked BTT, though the BTT markings were painted over.  A bit of alcohol cleared that up.  The touch display is "dumb" and controlled by the control board processor.  Interestingly, my D301 is populated with TMC2209s.  To get any further, I think I'll have to learn to debug with STLINK, but I'm probably 95% of the way to a working build of mainline Marlin.

ScottWarrick, 2022-12-21:
A few of us got Mingda D2s back in February-March 2020 and tried to plow this same ground. If I had seen this thread before, I could have saved you guys some time. 

At that time Mingda would only provide an older version of the source (their version 2.72) that did not match their binary (their version 2.82). So we were worrying from an even older source than here.

The MD-301 is a custom (proprietary) board that has a ribbon cable that attaches to a custom (proprietary screen). She screen is the ultimate issue. Their source is completely dependent upon extensibleUI. But the screen code is not where it should be but instead spread all over the place including inside parts of the main code. One guy in the group tried to replace the board and screen only to find that the clearance for the thin proprietary screen was so tight that he could find no screen to fit in that space. I found myself with a printer that was almost impossible to upgrade firmware and difficult to upgrade the hardware to then upgrade firmware. I was within my 30 days Amazon window so I returned it and gave an honest poor review.

A couple of months later Mingda sent me a Rock 3 Pro for free (long story). It is basically the same board, screen, and printer with a larger build surface. They also released this version of the source (their 2.82 binary). A couple of us looked at it and similarly realized they had taken the previous old version and added in most of the patches along with some custom hacks. That was the end for me. I decided I would either use the printer as is or let it collect dust.

Well, it is collecting dust. Thankfully I got it for free. Mingda took the Rock 3 series off it’s website, deleted all firmware and source downloads, and provides zero support. I recall when Kyon got his Mingda was coming out with its Magician series and overnight began pretending the Rock 3 series never existed.

The problem now is that this firmware is very behind. For example it will never have the input shaping that just came out. And hoo boy does it need it.

I was looking to see if anyone had gotten it into the main Marlin. This is the closest I have seen that still worked with the screen. I was able to get main Marlin to compile and onto the board without any screen support. Some people even decided to use Klipper on it with a Pi/screen which got around some of these issues. I did not want to invest anything into the printer including a Pi.

That being said, if someone wants me to test something on the printer I can fire it up and put firmware on there and see what happens.

I wish I had seen this thread a long time ago, I might could have saved you guys some time figuring all this out.

CommanderMoto, 2022-12-21:
(hilariously, I gave up on this attempt pretty quickly. The Magician is now the only printer I own which has been continuously in working order since acquisition)

ellensp, 2023-10-13:
A PR has been created, closing this FR

closed 2023-10-13 by ellensp

lscarneiro, 2023-10-17:
@ScottWarrick and @kyon03, would you folks be able to give your Mingda D2 a test on #26340?
No pressure, though.