MarlinFirmware/Marlin #21982: [BUG] SKR2 serial partially working

LawnMo, 2021-05-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a BTT SKR2 board and an esp12s (btt RRF Wifi module for SKR2).  

According to the motherboard documentation, there's a double jumper to select wether U-disk or Spi Wifi is enabled, and it mentions that "U-disk jumper makes the wifi module work _only_ on UART, _only_ on Marlin).  
Which is why I decided to give the wifi module a try, I breadboarded the thing with 10k ohms and a serial-to-usb adapter and properly flashed it with ESP3D 2.1.1 (also flashed the "spiffs size detection" sketch, it has 4mb and RX/TX on the esp12s is working).  
I then plugged it to the motherboard and adapted my config (serial_port{1/2} to 3, LCD screen doesn't show any M117 messages from the ESP (no IP address, nothing), but ESP3D terminal shows some serial messages from marlin :  
if I use an usb flashdrive, plugged/mounted on the board, ESP3D won't get any messages from marlin, as soon as I eject/unplug my flashdrive, ESP3D gets spammed by "waiting for media...".  
Which is when I tested different media settings, in case the flashdrive usb port has pins/timers/serial shared with the Wifi module header pins, when I use the onboard SD OR my 12864 SDcard reader OR multi_volume with `#define DEFAULT_VOLUME SV_SD_ONBOARD` I always get marlin's serial messages.  

I then decided to look at what the ESP_xxx pins were attributed to in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0_common.h#L516  

Only :  
```
#define ESP_WIFI_MODULE_RESET_PIN           PC14
#define ESP_WIFI_MODULE_GPIO0_PIN           PB10
```
seems to be used by marlin in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/shared/esp_wifi.cpp  
pin definition is missing ESP_WIFI_MODULE_GPIO2_PIN but that pin on the motherboard is physically hardwired to VCC so it's pulled up anyway (and the esp module boots/connect to wifi) so it doesn't look to be the issue.  

```
(ESP-RX)  PD8
(ESP-TX)  PD9
```
should be working fine, but since ESP-TX (sending commands to marlin) doesn't work, I tried a few things found online :  
- making ESP3D wait 10s (and 15s)  
- powered by usb with nothing attached but the ESP, powered by 12v
- with/without usb cable and pronterface attached
- changing the SKR2 and ESP baudrate (125k, 250k, mismatched baudrate shows weird symbols)
- uncommenting ESP_WIFI_MODULE_RESET_PIN  
- swapping SERIAL_PORT / PORT1 / PORT2 to -1, 1, 3 in all possible orders  
- filling a bug report to ESP3D (luc-github seems to think it's a marlin issue)  
- inverting jumpers from U-disk to Wifi, and retested most of the above points  

pins_debugging / M43 shows PD8/PD9 (ESP rx/tx pins) uses "USART1..3" (I forgot to save M43's output, I might rebuild and post it if that's relevant).  

SKR2 uses a script to create a generic platformio variant and I couldn't (didn't really try to) find its' pin map, which is where I got lost ^^.  

### Bug Timeline

new bug, SKR2 release

### Expected behavior

working RX/TX serial between ESP and marlin

### Actual behavior

ESP-RX / PD8 pin works (marlin serial messages shows in ESP3D's terminal), ESP-TX / PD9 pin doesn't.

### Steps to Reproduce

1. any config that uses a SKR2 board  
2. any SERIAL_PORTx set to 3  
3. ESP12s with ESP3d 2.1.1 on it (default pio build, 160mhz)
4. plug the ESP into the SKR2's ESP header  
5. no ESP-TX / esp3d can't send commands  

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

diy

### Electronics

BTT SKR2

### Add-ons

BTT RRF Wifi module (an ESP12s)

### Host Software

SD Card (headless)

### Additional information & file uploads

[cfg.zip](https://github.com/MarlinFirmware/Marlin/files/6538088/cfg.zip)

alex2501, 2021-05-30:
I got the same issue with this combination and found the root cause. There is a conflict with pin PB11 which is defined for USART3 in variant too.

You can change line 217 in file Marlin/buildroot/share/PlatformIO/variants/MARLIN_F4x7Vx/PeripheralPins.c:
From
`  {PB_11, USART3,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART3)},`
To
`  //{PB_11, USART3,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART3)},`

For me it is working, but don't know if this would effect any other boards using this variant and/or if this is a correct fix.

LawnMo, 2021-05-30:
Good catch! Thanks for sharing :)  

I might wait for the confirmation that it's safe to apply, but it's nice to know my board is not DOA.  
I guess it can be ifdef'd for SKR2 A&B so it doesn't affect other boards 👍

rhapsodyv, 2021-05-30:
I don’t know why all the peripheral pins are commented out. There’s no harm letting everything uncommented.

LawnMo, 2021-05-30:
> 
> 
> I don’t know why all the peripheral pins are commented out. There’s no harm letting everything uncommented.

Did you mean the opposite or do you mean alex2501's workaround is coincidental ?

rhapsodyv, 2021-05-30:
I mean that @alex2501 patch is right, but we should not needed it if the variant file was right. Someone commented those lines, but it should not be commented at first place.

LawnMo, 2021-07-30:
still happens.

thinkyhead, 2021-07-31:
I wonder if this could be related to #22181 …. In that case the MMU2 uses a serial port, and somehow this is messing with endstops.

LawnMo, 2021-08-13:
Didn't test it more than homing and checking temperatures but : https://github.com/MarlinFirmware/Marlin/commit/feb6d2887eee1b31713ef0ad665b166c8fe6e70b appears to fix it, thanks.

closed 2021-08-13 by LawnMo