MarlinFirmware/Marlin #21990: [FR] (feature summary)Dual y platform running independently

diver197, 2021-05-26:
### Is your feature request related to a problem? Please describe.

I want to configure a dual y platform that can run independently. When I go home, I fail. Y2 runs home first. When Y1 is at home, Y2 can go home normally. If Y1 is not at home, Y2 endstop is useless

### Describe the feature you want

Dual y platform running independently

DerAndere1, 2021-06-12:
Please discribe the desired feature in more detail. I am not sure if I understand correctly. With newest bugfix-2.0.x You can set up `#define LINEAR_AXES 4` and `#define AXIS4_NAME 'V'` and follow the instructions in comments above the option LINEAR_AXES. Then you would have axes XYZ and a V-axis that can be used as a secondary linear axis that is parallel to the Y axis with the corresponding G1 syntax G1 [Xxxxx] [Yyyyy] [Zzzzz] [Vvvvv] [Eeeee] [Ffffff]. Homing of the primary Y axis would be done with `G28 Y`, homing of the secondary axis parallel to the Y axis would be done with `G28 V`

diver197, 2021-06-16:
Compiling .pio\build\SJ_V1\libd5a\Wire\Wire.cpp.o
In file included from .pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.cpp:27:
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: variable or field 'SoftSerial_Handler' declared void
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
      |                                    timer_t
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:37:46: error: 'htim' was not declared in this scope; did you mean 'tm'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                              ^~~~
      |                                              tm
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.cpp:69:1: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
   69 | stimer_t SSTimerHandle;
      | ^~~~~~~~
      | timer_t
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.cpp: In function 'void HAL_softSerial_init()':
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.cpp:74:3: error: 'SSTimerHandle' was not declared in this scope
   74 |   SSTimerHandle.timer = SS_TIMER_DEV;
      |   ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
In file included from .pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial.h:28,
                 from .pio\libdeps\SJ_V1\SoftwareSerialM\SoftwareSerial.cpp:40:
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: variable or field 'SoftSerial_Handler' declared void
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
      |                                    timer_t
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:37:46: error: 'htim' was not declared in this scope; did you mean 'tm'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                              ^~~~
      |                                              tm
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:35:71: error: variable or field 'SoftSerial_Handler' declared void
   35 | #define HAL_SOFTSERIAL_TIMER_ISR() extern "C" void SoftSerial_Handler(stimer_t *htim)
      |                                                                       ^~~~~~~~
.pio\libdeps\SJ_V1\SoftwareSerialM\SoftwareSerial.cpp:218:1: note: in expansion of macro 'HAL_SOFTSERIAL_TIMER_ISR'
  218 | HAL_SOFTSERIAL_TIMER_ISR() {
      | ^~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\SJ_V1\SoftwareSerialM\HAL_softserial_STM32.h:35:71: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
   35 | #define HAL_SOFTSERIAL_TIMER_ISR() extern "C" void SoftSerial_Handler(stimer_t *htim)
      |                                                                       ^~~~~~~~
.pio\libdeps\SJ_V1\SoftwareSerialM\SoftwareSerial.cpp:218:1: note: in expansion of macro 'HAL_SOFTSERIAL_TIMER_ISR'
  218 | HAL_SOFTSERIAL_TIMER_ISR() {
      | ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\SJ_V1\lib1ff\SoftwareSerialM\HAL_softserial_STM32.cpp.o] Error 1
*** [.pio\build\SJ_V1\lib1ff\SoftwareSerialM\SoftwareSerial.cpp.o] Error 1

ellensp, 2021-06-16:
how about attaching some configs so we can look at this?

diver197, 2021-06-17:
The problem caused by the change of the compiled files is that the compiled scripts between different versions change too fast and cannot be synchronized

diver197, 2021-06-17:
> 请更详细地描述所需的功能。我不确定我是否理解正确。随着最新的漏洞修复-2.0.x的您可以设置`#define LINEAR_AXES 4`和`#define AXIS4_NAME 'V'`并按照上面的选项LINEAR_AXES在Configuration.h意见的说明。然后您将拥有 XYZ 轴和一个 V 轴，它们可以用作平行于 Y 轴的辅助线性轴，使用相应的 G1 语法 G1 [Xxxxx] [Yyyyy] [Zzzzz] [Vvvvv] [Eeeee] [Ffffff ]。主 Y 轴的`G28 Y`归位将通过 完成，平行于 Y 轴的辅助轴的归位将通过`G28 V`

G28 V

diver197, 2021-06-17:
> Please discribe the desired feature in more detail. I am not sure if I understand correctly. With newest bugfix-2.0.x You can set up `#define LINEAR_AXES 4` and `#define AXIS4_NAME 'V'` and follow the instructions in comments in Configuration.h above the option LINEAR_AXES. Then you would have axes XYZ and a V-axis that can be used as a secondary linear axis that is parallel to the Y axis with the corresponding G1 syntax G1 [Xxxxx] [Yyyyy] [Zzzzz] [Vvvvv] [Eeeee] [Ffffff]. Homing of the primary Y axis would be done with `G28 Y`, homing of the secondary axis parallel to the Y axis would be done with `G28 V`

G28，G28 V move ok 。‘G0 V100’or ‘G1 V100’  No move

closed 2021-06-17 by diver197