MarlinFirmware/Marlin #21991: [FR] Animated Bootscreen Example

AvanOsch, 2021-05-27:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

No

### Describe the feature you want

In Marlin 2.0.8.x there are new features for an animated bootscreen.
I believe this includes different duration for each animated image.
I can't find any documentation on how to use this.

Looking at this code, we should define our bootscreens differently: (./src/lcd/dogm/marlinui_DOGM.cpp)
```
      #if ENABLED(CUSTOM_BOOTSCREEN_ANIMATED)
        #if ENABLED(CUSTOM_BOOTSCREEN_ANIMATED_FRAME_TIME)
          const u8g_pgm_uint8_t * const bmp = (u8g_pgm_uint8_t*)pgm_read_ptr(&custom_bootscreen_animation[frame].bitmap);
        #else
          const u8g_pgm_uint8_t * const bmp = (u8g_pgm_uint8_t*)pgm_read_ptr(&custom_bootscreen_animation[frame]);
        #endif
      #else
        const u8g_pgm_uint8_t * const bmp = custom_start_bmp;
      #endif
```
Could someone elaborate?
(@thinkyhead maybe)

thisiskeithb, 2021-05-27:
See [this animated bootscreen example](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/AnimationExample).

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-05-27 by thisiskeithb

AvanOsch, 2021-05-28:
Thanks for your response.
However, the example you linked to is not using the newly added options.
I'll agree that this is not exactly a feature request, but the new functionality isn't documented or discussed anywhere...

In case someone comes here for the same reason, I looked through the code, learned about "typedef struct"s and came up with the following:
Instead of defining "custom_bootscreen_animation" as in the current examples, I did the following:
```
    const boot_frame_t custom_bootscreen_animation[] PROGMEM = {
        {custom_start_bmp, 2000}, {custom_start_bmp1, 100}, {custom_start_bmp2, 100},
        {custom_start_bmp3, 100}, {custom_start_bmp4, 100}, {custom_start_bmp5, 500}
    };
```
Not sure if this is the best way, but it works as expected.

I created a pull request with an adjusted example file.